solvespace/solvespace #430: Optimize GenerateAll to avoid doing unnecessary work

whitequark, 2019-05-23:
* It's not necessary to regenerate each entity when doing something simple like dragging a point in the very last group: https://github.com/Evil-Spirit/solvespace-master/commit/a5ffd9d182360e1be904aa35e2f9eb331783b211
  * It's not necessary to move actual entities (large) in IdList; it is possible to add a level of indirection so that only indexes (small) move: https://github.com/Evil-Spirit/solvespace-master/commit/9cf1a09bcdaa5a2bc99400d368dc1a8f43996e1d

Evil-Spirit, 2019-06-14:
I actually don't know what because of which commit, but my branch https://github.com/Evil-Spirit/solvespace-master/tree/user-param-proto very fast in comparsion with 2.3. Now I am drawing my brand-new house 
[image]
[image]

whitequark, 2019-06-14:
Very nice. Congratulations on the house :D

Evil-Spirit, 2019-06-14:
Drawing houses in reality isn't same thing as in SolveSpace ;)

Evil-Spirit, 2020-05-13:
Sorry for spaming:
[image]

[image]

[image]

[image]

ruevs, 2020-05-13:
Offtopic: Watched your video with the laser level - impressive precision - beautiful!

whitequark, 2020-05-16:
@Evil-Spirit That looks really nice!