solvespace/solvespace #431: Regression: out of bounds write on loading file

rpavlik, 2019-05-23:
### System information

SolveSpace version: master branch commit 43a59e212f68368db4d31748cdb66f8f90e3bf57

**Git bisect reports that cc107887e727fdc45186cca44e74129629f92de6 is the first broken commit.**

Operating system: Debian Buster

### Expected behavior

Should be able to open this file, just as I did earlier this week.

### Actual behavior

Crashes in DeleteAllTemporaries with a glibc error:

```
corrupted size vs. prev_size
fish: “bin/solvespace” terminated by signal SIGABRT (Abort)
```

web search suggested out of bounds write to blame.

### Additional information

This is the file that fails to open:

[screwdriver-sheath-with-taper-skinny-screwdriver.zip](https://github.com/solvespace/solvespace/files/3214236/screwdriver-sheath-with-taper-skinny-screwdriver.zip)

This is a closely-related one (its predecessor, since I'm using the "save as" method of version control for these) that does still open in master.

[screwdriver-sheath-with-taper.zip](https://github.com/solvespace/solvespace/files/3214240/screwdriver-sheath-with-taper.zip)

whitequark, 2019-05-23:
What about:
```patch
diff --git a/src/mesh.cpp b/src/mesh.cpp
index 7767901..ba3c116 100644
--- a/src/mesh.cpp
+++ b/src/mesh.cpp
@@ -194,7 +194,7 @@ void SMesh::Simplify(int start) {
                         conv[WRAP((j+1), convc)] = c;
                     } else if(bDot > 0 && dDot > 0) {
                         // conv[j] is unchanged, conv[j+1] goes to [j+2]
-                        std::move_backward(conv+j+1, conv+convc, conv+j+2);
+                        std::move_backward(conv+j+1, conv+convc-1, conv+j+2);
                         conv[j+1] = c;
                         convc++;
                     } else {
```

rpavlik, 2019-05-23:
That crashes sooner, on that line:

[image]

whitequark, 2019-05-23:
Uh. This?
```diff
diff --git a/src/mesh.cpp b/src/mesh.cpp
index 7767901..8ab27e0 100644
--- a/src/mesh.cpp
+++ b/src/mesh.cpp
@@ -186,7 +186,7 @@ void SMesh::Simplify(int start) {
                     if(fabs(bDot) < LENGTH_EPS && fabs(dDot) < LENGTH_EPS) {
                         conv[WRAP((j+1), convc)] = c;
                         // and remove the vertex at j, which is a dup
-                        std::move(conv+j+1, conv+convc, conv+(convc-1));
+                        std::move(conv+j+1, conv+convc, conv+j);
                         convc--;
                     } else if(fabs(bDot) < LENGTH_EPS && dDot > 0) {
                         conv[j] = c;
```

rpavlik, 2019-05-24:
Hmm, nope, that's back to the original crash.

```
#0  0x00007ffff66b87bb in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff66a3535 in __GI_abort () at abort.c:79
#2  0x00007ffff66fa508 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff680528d "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff6700c1a in malloc_printerr (str=str@entry=0x7ffff6803376 "corrupted size vs. prev_size") at malloc.c:5341
#4  0x00007ffff6700e74 in malloc_consolidate (av=av@entry=0x7ffff683cc40 <main_arena>) at malloc.c:4488
#5  0x00007ffff670279a in _int_free (av=0x7ffff683cc40 <main_arena>, p=0x555556f0c640, have_lock=<optimized out>) at malloc.c:4392
#6  0x000055555560e989 in SolveSpace::FreeAllTemporary() () at ../src/platform/utilunix.cpp:63
#7  0x0000555555665661 in SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate, bool, bool) (this=this@entry=0x5555558a3d40 <SolveSpace::SS>, type=type@entry=SolveSpace::SolveSpaceUI::Generate::ALL, andFindFree=andFindFree@entry=false, genForBBox=genForBBox@entry=false) at ../src/generate.cpp:359
#8  0x00005555555eb1e1 in SolveSpace::SolveSpaceUI::AfterNewFile() (this=0x5555558a3d40 <SolveSpace::SS>) at ../src/solvespace.cpp:387
#9  0x00005555555ee7db in SolveSpace::SolveSpaceUI::Load(SolveSpace::Platform::Path const&) (this=0x5555558a3d40 <SolveSpace::SS>, filename=...) at ../src/solvespace.cpp:188
#10 0x00007ffff7cc7e6e in Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#11 0x00007ffff632dc7d in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007ffff63414ec in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007ffff634a25e in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#14 0x00007ffff634a91f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#15 0x00007ffff752988a in gtk_widget_activate () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#16 0x00007ffff73fca86 in gtk_menu_shell_activate_item () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#17 0x00007ffff73fcd23 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#18 0x00007ffff7579274 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#19 0x00007ffff632deb6 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007ffff6349d14 in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007ffff634a91f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#22 0x00007ffff7527324 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#23 0x00007ffff73e7976 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#24 0x00007ffff73e9a83 in gtk_main_do_event () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#25 0x00007ffff70eb465 in  () at /usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#26 0x00007ffff711c112 in  () at /usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#27 0x00007ffff6f1af2e in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007ffff6f1b1c8 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff6f1b4c2 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff73e8b15 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#31 0x00005555555e521d in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at ../src/platform/entrygui.cpp:28
```

whitequark, 2019-05-24:
Oh, oops. Fixed in 9faa7cb0.

closed 2019-05-24 by whitequark