solvespace/solvespace #437: Failed to wrap object of type 'GtkWindow'

Symbian9, 2019-05-31:
### System information

SolveSpace version: 3.0~9ac55f39

Operating system: Debian 9.x Stretch (amd64)

### Expected behavior

SolveSpace should work without issues.

### Actual behavior

After launch SolveSpace it print too many  `glibmm-CRITICAL` and `glibmm-WARNING` messages into terminal output.

### Additional information

Terminal Log:

<details>

```
$ solvespace
SolveSpace!

Missing (absent) translation for 'Show &Underconstrained Points'
Missing (absent) translation for dialog'Do you want to save the changes you made to the new sketch?'
Missing (absent) translation for dialog'Your changes will be lost if you don't save them.'
Missing (absent) translation for button'&Save'
Missing (absent) translation for button'Do&n't Save'
Missing (absent) translation for button'&Cancel'

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.645: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.647: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.648: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.648: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.648: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.648: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.649: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.649: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.650: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.650: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.650: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.650: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.651: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.651: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.651: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.652: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.652: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.653: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.653: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.653: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.
[... 414 lines cut ...]

(solvespace:3535): glibmm-WARNING **: 18:43:01.702: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.702: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.702: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.702: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.703: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.703: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.703: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:3535): glibmm-CRITICAL **: 18:43:01.703: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:3535): glibmm-WARNING **: 18:43:01.703: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.
```

</details>

### REFERENCE

- http://solvespace.com/forum.pl?action=viewthread&parent=2697&tt=1559114394

whitequark, 2019-05-31:
It's actually a gtkmm bug: https://gitlab.gnome.org/GNOME/gtkmm/issues/22.

whitequark, 2019-05-31:
Fixed in 3d80062b.

closed 2019-05-31 by whitequark