solvespace/solvespace #438: Missing (absent) translation

Symbian9, 2019-05-31:
### System information

SolveSpace version: 3.0~ https://github.com/solvespace/solvespace/commit/9ac55f392aeac6839e1ad2252848f520f1420939

Operating system: Debian 9.x Stretch (x86_64)

### Expected behavior

SolveSpace should start & work without issues.

### Actual behavior

After launch SolveSpace it print in terminal output messages about `Missing (absent) translation ...`

### Additional information

Terminal Log:

<details>

```
$ solvespace
SolveSpace!

Missing (absent) translation for 'Show &Underconstrained Points'
Missing (absent) translation for dialog'Do you want to save the changes you made to the new sketch?'
Missing (absent) translation for dialog'Your changes will be lost if you don't save them.'
Missing (absent) translation for button'&Save'
Missing (absent) translation for button'Do&n't Save'
Missing (absent) translation for button'&Cancel'

...
```

</details>

whitequark, 2019-05-31:
That's not a bug. That's a missing translation, exactly as the message says.

closed 2019-05-31 by whitequark

Symbian9, 2019-05-31:
If so, could this message include also locale codes of translation (e.g. `ru`, `uk`, etc.) where those messages missing?

Also, if I not switch locale (use default *English*), think, those messages should not be printed as English is base language. Or we should create `en_UK` and `en_US` locales for SolveSpace?

whitequark, 2019-05-31:
> If so, could this message include also locale codes of translation (e.g. `ru`, `uk`, etc.) where those messages missing?

The messages are missing from the locale you've selected.

> Also, if I not switch locale (use default _English_), think, those messages should not be printed as English is base language.

These messages are printed for English if the translation template (`.pot` file) has not been updated. This file should only be updated before requesting translators to update everything for a new release, so it'll be before a release. You can safely ignore them otherwise.