solvespace/solvespace #440: Objects flip to the opposite side constraints

ad8e, 2019-06-12:
### System information

SolveSpace version: 3.0~5df53fc5
Operating system: Ubuntu 19.04 in VMWare

### Expected behavior

When I drag points quickly or set constraints that move points a moderate distance, sometimes objects tunnel through their constraints to a different local solution (for example, if a point-line distance is specified, the point will flip to the opposite side of the line). My hope is that this occurrence is reduced in frequency.

### Additional information

If the numerical solver reuses previous state and flows along a gradient to the new state, maybe smaller steps would help. Alternatively, if a point is moved a large distance, a line can be drawn between the two positions and then the solution incrementally iterated along this line. This creates a smooth path between the start and end position, preventing discontinuous jumps.

https://github.com/solvespace/solvespace/issues/76 is related and the preferred solution. Mine is just a bandaid.

whitequark, 2019-06-12:
I don't think a band-aid that makes dragging significantly slower (and it's already not very fast) is desirable.

closed 2019-06-12 by whitequark

Evil-Spirit, 2019-06-13:
https://github.com/NoteCAD/NoteCAD already uses chirality-preserving equations where it is possible, so anyone can look into.