solvespace/solvespace #443: Request option for comments to be constrained/tied to other objects

philtipping, 2019-06-22:
### System information
SolveSpace version: 3.0~cb0fdb1c
Operating system: lubuntu 18.04

Request option for comments to be constrained/tied to other objects, and/or to each other.
I use lots of comments to label construction entities, but they all go out of alignment if an entity moves. If this is not easy to implement, can comments at least be dragged en masse when multiple comments are selected?

Have to add, am really impressed by solvespace - super stable and very slick & easy user interface - a breath of fresh air compared with other 'windows-heavy' programs. Kudos to the creator and all the contributors/maintainers.

Evil-Spirit, 2019-06-23:
Also, comments should be selected by marquee like the other constraints.

whitequark, 2019-06-23:
IIRC comments not being selected is deliberate.

phkahler, 2021-05-12:
See PR #1035. Selecting a point prior to adding a comment constraint ties the comment to the point. It's not clear how they should attach to other entities, but everything has at least one point ;-)

Feel free to reopen this or create another issue if this isn't good enough.

closed 2021-05-12 by phkahler