solvespace/solvespace #444: Request for constraints on font types and sizes

philtipping, 2019-07-02:
### System information
SolveSpace version: 3.0~cb0fdb1c
Operating system: lubuntu 18.04
### Additional information
I'd like to be able to control the type & size of multiple ttf-text items by constraining them to a master reference. I can work round the size requirement by using a construction line between two vertical points on the item, but can't see a way of handling the font type.

whitequark, 2019-07-02:
Reasonable, but I'm not sure how that fits into the current file format, so it'll need investigation.

philtipping, 2019-07-02:
An alternative which doesn't need constraints would be the ability to assign a font to all selected items.

Symbian9, 2019-07-02:
> An alternative which doesn't need constraints would be the ability to assign a font to all selected items.

Think, adding "Font" property to styles would be much better for this.

philtipping, 2019-07-02:
Yes, that would work too (for me).

whitequark, 2019-07-03:
Honestly, that's not any better with regards to the file format, and probably just as complex to implement. So the main question here is what UI should be exposed.