solvespace/solvespace #445: Opening file containing asemblies says: The linked file "" is not present.

philtipping, 2019-07-06:
### System information
SolveSpace version: 3.0~02d7f0c0
Operating system: lubuntu 18.04

### Expected behavior
Assemblies should load
### Actual behavior
Error message: The linked file "" is not present.
File contains multiple assemblies, although they are not nested. I located one of them manually and the file then loaded ok, but every assembly referred to the same part.
### Additional information
Found a work-round by editing the main slvs file.
Assemblies are on lines such as:
Group.impFileRel=../Parts/Toggle-Switch-3P.slvs
but some entries had an odd path name, e.g.
Group.impFileRel=../../../../../../home/phil/Documents/CADDesigns/Parts/Toggle-Switch-3P.slvs
I edited all longs paths to conform to the short ones, and the file now loads ok with no errors, and all assemblies correct.
The folder 'Documents' is on an ntfs partition which is mounted manually after lubuntu boots up, if that helps.

whitequark, 2019-07-06:
That's a bit troublesome. Do you think you can figure out a way to reproduce this?

philtipping, 2019-07-07:
Tried loading the original failing file today, and it now works!
Tried creating a new file with a new linked file, and this was ok, but noticed the stored pathname used the long format, so both short & long formats must be valid alternatives to the linked file. 
Checked previous versions of the original file which load ok, and these also contain a mixture of short and long pathnames, so looks like the pathname format was not the cause of the error message, even though editing them seemed to fix the problem yesterday. 
Bottom line is I can't reproduce this at the moment, but if you can add more diagnostic info to the error message (e.g. Group.name) I could maybe pinpoint the rogue part if it happens again.
Thank you.

phkahler, 2020-09-22:
I've added a debug message. This will print the name of a group whose associated file can't be found. In the issue report here, the dialog that popped up showed no file name so that was not helpful. This message will appear in the terminal window, so if you don't have one you need to open one and run SolveSpace from there.

Going to remove the 3.0 milestone from this but leave it open for when someone has it happen again.