solvespace/solvespace #449: Calculate volume of the current group / surface area of selected faces

Evil-Spirit, 2019-07-11:
This can be useful (I use this to calculate concrete volume).
https://github.com/Evil-Spirit/solvespace-master/commit/ef85e9e81de4dfa26356b264f57e76a18bd98834

phkahler, 2019-07-11:
It may be a stretch, but could these values be used in the solver? I saw someone using a different CAD use a volume constraint to set the height of a container. I doubt that's really feasible here, just wanted to share the thought.

whitequark, 2019-07-11:
That function is not analytical, is it?

ruevs, 2019-07-11:
That's an interesting idea. But apart from the Volume(dimensions) function not being analytical, the volume is calculated on the triangle mesh - which is certain to cause "noise" in the value.

phkahler, 2019-07-11:
@whitequark That was a dumb question from me. Of course volume is computed numerically from the shell/mesh. While there may (theoretically) be a way to twiddle free parameters to get numeric derivatives, that isn't a thing in SS.

@ruevs Good point. A simple triangle mesh shouldn't be a problem but when the number of triangles changes there would be a discontinuity in the calculated volume.

phkahler, 2019-07-12:
Do the triangles have a consistent ordering? Meaning ABC is always CCW or CW when viewed from the outside? If so, the volume can be calculated by summing a.Cross(b).Dot(c)/6 for all the triangles. That would be a lot less code.

Evil-Spirit, 2019-07-13:
@phkahler look at the STriangle::SignedVolume used in GetCenterOfMass(). Code of @jwesthues leaved intentionally since I have no ability to test new Volume calculation funtions.

Evil-Spirit, 2019-07-13:
BTW can someone explain me why I wrote div by 4 instead of div by 3 here:
https://github.com/solvespace/solvespace/blob/master/src/mesh.cpp#L352
I can't remember why, so this is probably a typo (since 3 and 4 close to each other).

phkahler, 2019-07-13:
@Evil-Spirit The division by 4 is because you're calculating the centre of a tetrahedron formed by the 3 triangle verticies and the origin. No need to add in the vector (0,0,0) but you still need to divide by 4 since there really are 4.

So to my point, you can just sum the STriangle::SignedVolume() of each triangle to get the total volume. Its implementation is exactly what I indicated in the previous comment.

This is explained in the following stack overflow question:
[general-formula-to-calculate-polyhedron-volume](https://stackoverflow.com/questions/1838401/general-formula-to-calculate-polyhedron-volume/1849746#1849746)

Evil-Spirit, 2019-07-13:
> So to my point, you can just sum the STriangle::SignedVolume() of each triangle to get the total volume. 
> Its implementation is exactly what I indicated in the previous comment.

Yes this will be good. You are welcome to contribute!

phkahler, 2019-09-10:
@whitequark  Does anything else need to be done for this to get in?

whitequark, 2019-09-11:
@phkahler Yes--SMesh::CalculateSurfaceArea shouldn't be a function on the mesh. As usual with commits that combine random refactoring that wasn't even necessary with adding a feature, it takes a while to review and merge.

Evil-Spirit, 2019-09-11:
@whitequark please, write more arguments why SMesh::CalculateSurfaceArea should be implemented in different way.
@phkahler it need to be merged by hands because I use old branch. Sorry, I have no to time to rebase.

whitequark, 2019-09-11:
> please, write more arguments why SMesh::CalculateSurfaceArea should be implemented in different way.

Sorry, I was wrong because I misread the code. I'll merge it soon.

whitequark, 2019-09-11:
@Evil-Spirit Thank you for this contribution. Merged as 915f55aabc201b79abe5f3b70ab9b95ce52a6749 and 7f9117b2bf0636a594eb1f1ba76b9dbe0e8a4e83.

closed 2019-09-11 by whitequark