solvespace/solvespace #450: Request to move coordinate axes from under the toolbar

philtipping, 2019-07-12:
### System information

SolveSpace version: 3.0~02d7f0c0

Operating system: lubuntu 18.04 on Dell Vostro laptop

### Additional information
Minor cosmetic request. The coordinate axes at the bottom-left of the main screen are partially hidden by the toolbar. There is room on my screen layout to move them lower, or maybe move the toolbar higher - see screenshot, but if this causes problems, a shortcut key for toggling the toolbar on/off would be a compromise. I know the axes are replicated in the centre but these can become difficult to select with a complicated diagram.
Thank you.
[image]

whitequark, 2019-07-12:
Commit 02a9a1b9 moves the axes to the right if the window is not tall enough.
[image]

closed 2019-07-12 by whitequark

whitequark, 2019-07-12:
By the way there is a `View` → `Show Toolbar` option.

philtipping, 2019-07-12:
Perfect thanks. 
btw View,Toolbar is what I'm using now but this just seemed cumbersome when you're used to using shortcuts :)