solvespace/solvespace #451: How to enable linkage to system-installed libs instead of bundled ones.

Petross404, 2019-07-23:
Basically the title. I want to use system installed on my gentoo dev-libs/flatbuffers. Running

``` bash
 build/ $ cmake-gui -DCMAKE_DISABLE_FIND_PACKAGE_FLATBUFFERS=ON ..
```
doesn't work.

Also I [can't find](https://github.com/solvespace/solvespace/blob/master/CMakeLists.txt#L113) a way to disable it somehow.
``` cmake
message(STATUS "Using in-tree flatbuffers")
set(FLATBUFFERS_BUILD_FLATLIB ON CACHE BOOL "")
set(FLATBUFFERS_BUILD_FLATC ON CACHE BOOL "")
set(FLATBUFFERS_BUILD_FLATHASH OFF CACHE BOOL "")
set(FLATBUFFERS_BUILD_TESTS OFF CACHE BOOL "")
add_subdirectory(extlib/flatbuffers EXCLUDE_FROM_ALL)
```

What am I missing here? Thank you.

whitequark, 2019-07-23:
This isn't currently supported, mostly because Debian does not ship flatbuffers, and most Linux users of SolveSpace are on Debian. Is there any particular reason you want it? Flatbuffers are currently only used for writing files, so using a potentially outdated version of them isn't a concern.

Petross404, 2019-07-23:
Thank you for your explanation. Yes, the reason I am looking at this is for gentoo packaging policies. An ebuild for solvespace already exists on the offical portage tree, but there are these policies that want system - installed libraries to be strongly preffered over bundled and avoid [auto-magic dependencies](https://wiki.gentoo.org/wiki/Project:Quality_Assurance/Automagic_dependencies).

But anyway, I am not a gentoo dev to take responsibility and as I mentioned an ebuild already exists. So unless the maintainer or a dev approaches you for this, I think you can ignore this issue.

All the best, 
Petros

whitequark, 2019-07-23:
Flatbuffers is not an automagic dependency as it is a hard requirement (which is also the reason for bundling it). If a Gentoo packager wants me to make it possible to use system flatbuffers they can comment here and I will reopen.

closed 2019-07-23 by whitequark

Petross404, 2019-07-23:
Yes, it isn't automagic dep to be honest. 

All the best,
Petros