solvespace/solvespace #454: gives insane dimension between 2 obviously-constrained points

libweirdness, 2019-08-06:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)
SolveSpace "2.3~"
( from Synaptic:
1.2+repack1-3build1 )

Operating system: (e.g. Debian testing)
UbuntuStudio.org 19.04,
updated to 2019-August-05

### Expected behavior
Was setting in place framework of Construction/Geometry lines,
to use for placing workplanes on
( having a slice through the model every 1', in this set ),
so I can try designing boats on this software,
got lots of consecutive 12" segments set in X & Y axes,
set new workplane to get the Z axis,
( g003 )
put a series of 'em on the Z axis,
constrained the 1st one to 12", the rest to be equal to the previous one,
distance between the 1st 2 points ( ends of the 1st segment )
*visibly-constrained* to 12"
EXPECT to get distance,
in the Property Browser window,
same 12"

### Actual behavior
Property Browser instead is showing as d =  168.428"
No idea why,
but it means my using this for solving stuff is borked,
because it can't be trusted.
( I'm arithmetic-disabled, so I NEED something to un-screw-up arithmetic for me,
this app was so wonderful until now )

Maybe the Git version that is current, isn't borked,
but Ubuntu has got to get a fixed version instead of what they got,
then...

What should have happened?
I expected that the behaviour would be the same in the XZ plane as in the XY plane,
that if I selected 2 points 12" apart, then the Property Browser would show 12"
I wanted to make enough of them so that there was 336" from end to end of the segments,
then I would have enough points to put workplane/slices on, to get the hang of designing a small sailboat.

What actually happened?
Instead I got insanity.
"in plane: g003-sketch-in-plane
and all those "12" segments, end-to-end, all vertical, from Origin up,
& the distance is misreported hugely.

Also, there shouldn't be any fractional-component in the distance
( a previous attempt to make that work had ended up with the segments NOT in the plane,
so I just group selected 'em all, deleted 'em, & went back to working in plane,
trying again.

Maybe it kept the scrambling from that mistake?

Will have to try that, beginning all the way from nothing, no mistakes in my sequence,
to see if that fixes it: if mistakes get kept, "under the hood" & prevent a file from working/solving correctly forevermore... )

### Additional information

1stly, TO EVERYONE WHO HELPED MAKE THIS APP,
*THANK YOU SO VERY VERY MUCH*
Dear Allgod, it is SO good to make my mistakes in digital, instead of wrecking everything in physical-implimentation, due to some brain deficits.
Solving everything in digital means being ABLE to implement designs, instead of never succeeding!

If I can do some experiment to help solve what is going on, please just ask.

If it is something that is wrong with my system
( how do I get a dependency-tree, to see what stuff it is incorporating?
maybe glibc is borked or something? )
it is the UbuntuStudio.org variant of Ubuntu 19.04, updated pretty-much daily,
but I have added tons of stuff in to try them, to discover what is worth relying-on
( do NOT rely on the KNotes that comes with this:
it doesn't save particularly recently, throwing away info from your notes )

Also, there is a Radeon driver problem in the system
( every few seconds the system stops for a few seconds,
annoying as hell, but soon will have a different computer
dmesg gives this bit:
"[10451.800181] radeon 0000:01:00.0: WB enabled
[10451.800184] radeon 0000:01:00.0: fence driver on ring 0 use gpu addr 0x0000000040000c00 and cpu addr 0x0000000069722d90
[10451.800185] radeon 0000:01:00.0: fence driver on ring 1 use gpu addr 0x0000000040000c04 and cpu addr 0x00000000a6a0af49
[10451.800186] radeon 0000:01:00.0: fence driver on ring 2 use gpu addr 0x0000000040000c08 and cpu addr 0x00000000f0997ba7
[10451.800187] radeon 0000:01:00.0: fence driver on ring 3 use gpu addr 0x0000000040000c0c and cpu addr 0x00000000541b138e
[10451.800188] radeon 0000:01:00.0: fence driver on ring 4 use gpu addr 0x0000000040000c10 and cpu addr 0x00000000a6fe2469
[10451.800623] radeon 0000:01:00.0: fence driver on ring 5 use gpu addr 0x0000000000075a18 and cpu addr 0x00000000857fe22f
[10451.902128] radeon 0000:01:00.0: failed VCE resume (-110)."

I don't know if THAT scrambles something that SolveSpace relies on, either
( dual GPU, an Intel onboard junker that is low power,
the Radeon,
& me looking at this on an external screen through a DisplayLink adapter on a non-tiny screen )

-sigh- maybe this bug cannot be reproduced on any normal computer, possible,
but maybe it can.

Anyways, to anyone who can help, thank you, & again,
thank youse for making this app in the 1st place!

Salut, Namaste, & Kaizen!

( :

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.
[Template__Boat_Stations.slvs.zip](https://github.com/solvespace/solvespace/files/3470217/Template__Boat_Stations.slvs.zip)

phkahler, 2019-08-24:
This appears to be a point selection issue. I opened the file and could not initially reproduce the problem. However, if you select the two points shown here, the distance is 168.428.

[image]

If @libweirdness was looking at the XZ plane and somehow selected the nearest point instead of the one at the origin, it would measure that distance from the first point up the Z axis. Using middle-click-drag to rotate the view would help to show which points are being measured.

In short, distances are measured in 3D so be careful what is selected when things are on top of each other.

whitequark, 2019-09-02:
Duplicate of #173, I believe.

closed 2019-09-02 by whitequark