solvespace/solvespace #458: Revolve shouldn't be possible for 3D groups

rpavlik, 2019-08-14:
### System information

SolveSpace version: 3.0~97c8cb7d

Operating system: Debian buster

### Expected behavior

I tried to use the new "Revolve" feature, initially trying to sweep a solid. It appears that is not supported (only sweeping a 2D sketch), so it shouldn't have let me do it.

### Actual behavior

It let me do it, but the result didn't make any sense.

phkahler, 2019-08-14:
Lathe isn't very useful for solids either.

rpavlik, 2019-08-19:
Probably pop up the same kind of window that I get when trying to make a new workspace group without selecting an origin, etc. (or just grey out that option?)

phkahler, 2019-08-22:
The check for that would go in Group::MenuGroup() under the cases for those group types. Question on Localization, is it OK if we use the exact same string for the error message in multiple places in the code? If so, should we?

The test would be similar to the one used in Extrude groups and should be in Revolve and Helix groups too. It occurs to me that in weird cases, doing this with Lathe might create circles that are relevant to someone somewhere.

whitequark, 2019-08-22:
> Question on Localization, is it OK if we use the exact same string for the error message in multiple places in the code? If so, should we?

Completely identical messages get grouped into a single group by gettext so that's fine.

> It occurs to me that in weird cases, doing this with Lathe might create circles that are relevant to someone somewhere.

Fortunately, existing files with such strange constructs will continue to work correctly. And if someone complains and it turns out we missed an important use case, we can always revert.

phkahler, 2019-08-25:
Implemented Fix in commit 2dd50d0 of PR #466

closed 2019-08-26 by whitequark