solvespace/solvespace #462: Double-free detected with ASan

rpavlik, 2019-08-20:
### System information

SolveSpace version: https://github.com/rpavlik/solvespace/commit/1d24d526fe372483ef49ec5e72b936406c4e90a6

Operating system: Debian buster

### Expected behavior

Build my "sanitize" branch with some recent clang. https://github.com/rpavlik/solvespace/tree/sanitize

Run `env ASAN_OPTIONS=new_delete_type_mismatch=0 build/bin/solvespace`, open a file, and switch the active group to group 5 - "lathe". It shouldn't crash/halt with an error.

(Note that ASan option is required to avoid breaking right away on something related to sigc++.)

### Actual behavior

It halts with the error below, rather than show the lathe as the active group.

```
=================================================================
==28310==ERROR: AddressSanitizer: attempting double-free on 0x620000160080 in thread T0:
    #0 0x6ce7e2 in free (/home/ryan/src/third-party/solvespace/build/bin/solvespace+0x6ce7e2)
    #1 0x8584b0 in SolveSpace::List<SolveSpace::STrimBy>::Clear() /home/ryan/src/third-party/solvespace/build/../src/dsc.h:285:18
    #2 0x869357 in SolveSpace::SShell::Clear() /home/ryan/src/third-party/solvespace/build/../src/srf/surface.cpp:1064:12
    #3 0x7e16cf in SolveSpace::Group::GenerateShellAndMesh() /home/ryan/src/third-party/solvespace/build/../src/groupmesh.cpp:188:15
    #4 0x7cf5b6 in SolveSpace::SolveSpaceUI::GenerateAll(SolveSpace::SolveSpaceUI::Generate, bool, bool) /home/ryan/src/third-party/solvespace/build/../src/generate.cpp:270:24
    #5 0x71ef7b in void std::__invoke_impl<void, void (SolveSpace::SolveSpaceUI::*&)(SolveSpace::SolveSpaceUI::Generate, bool, bool), SolveSpace::SolveSpaceUI*&, SolveSpace::SolveSpaceUI::Generate&, bool&, bool&>(std::__invoke_memfun_deref, void (SolveSpace::SolveSpaceUI::*&)(SolveSpace::SolveSpaceUI::Generate, bool, bool), SolveSpace::SolveSpaceUI*&, SolveSpace::SolveSpaceUI::Generate&, bool&, bool&) /usr/bin/../lib/gcc/x86_64-linux-gnu/8/../../../../include/c++/8/bits/invoke.h:73:14
    #6 0x71edf0 in std::__invoke_result<void (SolveSpace::SolveSpaceUI::*&)(SolveSpace::SolveSpaceUI::Generate, bool, bool), SolveSpace::SolveSpaceUI*&, SolveSpace::SolveSpaceUI::Generate&, bool&, bool&>::type std::__invoke<void (SolveSpace::SolveSpaceUI::*&)(SolveSpace::SolveSpaceUI::Generate, bool, bool), SolveSpace::SolveSpaceUI*&, SolveSpace::SolveSpaceUI::Generate&, bool&, bool&>(void (SolveSpace::SolveSpaceUI::*&)(SolveSpace::SolveSpaceUI::Generate, bool, bool), SolveSpace::SolveSpaceUI*&, SolveSpace::SolveSpaceUI::Generate&, bool&, bool&) /usr/bin/../lib/gcc/x86_64-linux-gnu/8/../../../../include/c++/8/bits/invoke.h:95:14
    #7 0x71ed07 in void std::_Bind<void (SolveSpace::SolveSpaceUI::* (SolveSpace::SolveSpaceUI*, SolveSpace::SolveSpaceUI::Generate, bool, bool))(SolveSpace::SolveSpaceUI::Generate, bool, bool)>::__call<void, 0ul, 1ul, 2ul, 3ul>(std::tuple<>&&, std::_Index_tuple<0ul, 1ul, 2ul, 3ul>) /usr/bin/../lib/gcc/x86_64-linux-gnu/8/../../../../include/c++/8/functional:400:11
    #8 0x71eb3d in void std::_Bind<void (SolveSpace::SolveSpaceUI::* (SolveSpace::SolveSpaceUI*, SolveSpace::SolveSpaceUI::Generate, bool, bool))(SolveSpace::SolveSpaceUI::Generate, bool, bool)>::operator()<void>() /usr/bin/../lib/gcc/x86_64-linux-gnu/8/../../../../include/c++/8/functional:482:17
    #9 0x71e950 in std::_Function_handler<void (), std::_Bind<void (SolveSpace::SolveSpaceUI::* (SolveSpace::SolveSpaceUI*, SolveSpace::SolveSpaceUI::Generate, bool, bool))(SolveSpace::SolveSpaceUI::Generate, bool, bool)> >::_M_invoke(std::_Any_data const&) /usr/bin/../lib/gcc/x86_64-linux-gnu/8/../../../../include/c++/8/bits/std_function.h:297:2
    #10 0x75f85b in std::function<void ()>::operator()() const /usr/bin/../lib/gcc/x86_64-linux-gnu/8/../../../../include/c++/8/bits/std_function.h:687:14
    #11 0x75f810 in SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int)::'lambda'()::operator()() const /home/ryan/src/third-party/solvespace/build/../src/platform/guigtk.cpp:217:17
    #12 0x75f7d8 in sigc::adaptor_functor<SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int)::'lambda'()>::operator()() const /usr/include/sigc++-2.0/sigc++/adaptors/adaptor_trait.h:256:12
    #13 0x75f7cc in sigc::internal::slot_call0<SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int)::'lambda'(), bool>::call_it(sigc::internal::slot_rep*) /usr/include/sigc++-2.0/sigc++/functors/slot.h:136:14
    #14 0x7f7b62881b11  (/usr/lib/x86_64-linux-gnu/libglibmm-2.4.so.1+0x5ab11)
    #15 0x7f7b6274d862  (/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4e862)
    #16 0x7f7b6274cdd7 in g_main_context_dispatch (/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4ddd7)
    #17 0x7f7b6274d1c7  (/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4e1c7)
    #18 0x7f7b6274d4c1 in g_main_loop_run (/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4e4c1)
    #19 0x7f7b62c1ab14 in gtk_main (/usr/lib/x86_64-linux-gnu/libgtk-3.so.0+0x23ab14)
    #20 0x75a968 in SolveSpace::Platform::RunGui() /home/ryan/src/third-party/solvespace/build/../src/platform/guigtk.cpp:1474:5
    #21 0x758c46 in main /home/ryan/src/third-party/solvespace/build/../src/platform/entrygui.cpp:28:5
    #22 0x7f7b61ea709a in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2409a)
    #23 0x6231b9 in _start (/home/ryan/src/third-party/solvespace/build/bin/solvespace+0x6231b9)

0x620000160080 is located 0 bytes inside of 3584-byte region [0x620000160080,0x620000160e80)
freed by thread T0 here:
    #0 0x6ce7e2 in free (/home/ryan/src/third-party/solvespace/build/bin/solvespace+0x6ce7e2)
    #1 0x8584b0 in SolveSpace::List<SolveSpace::STrimBy>::Clear() /home/ryan/src/third-party/solvespace/build/../src/dsc.h:285:18

previously allocated by thread T0 here:
    #0 0x6ceb63 in __interceptor_malloc (/home/ryan/src/third-party/solvespace/build/bin/solvespace+0x6ceb63)
    #1 0x77e7e5 in SolveSpace::MemAlloc(unsigned long) /home/ryan/src/third-party/solvespace/build/../src/platform/utilunix.cpp:69:15

SUMMARY: AddressSanitizer: double-free (/home/ryan/src/third-party/solvespace/build/bin/solvespace+0x6ce7e2) in free
==28310==ABORTING
```

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2019-08-20:
1. Do you think you can bisect it?
2. Does it reproduce with our testsuite?
3. Do you think you can add asan to the CI matrix? I think I tried it before and it didn't work because Travis' Ubuntu was too old or something, but it should be OK now.

rpavlik, 2019-08-20:
Well, it doesn't happen with the test suite.

The trouble with adding it to CI is that the dependencies aren't perfect, as in, the q3d header generator threw some error (I don't remember if it was memory or address sanitizer) pretty early on before it even finished compiling. That's why I had to hack up the cmake script.

At some point I probably can do this, though.

rpavlik, 2019-08-20:
ok, I'm the culprit - it was that "simplify6" branch you just merged.

PR #464 (2019-08-20): Revert "IdList::RemoveTagged switch to std::remove_if from iteration.…
… NFC."

This reverts commit 0bb6a348e393f460c0b1bf6ab04b4bae41102688.

Fixes #462
closed 2019-08-20 by whitequark