solvespace/solvespace #465: Replace struct "Revolved"

phkahler, 2019-08-22:
In surface .cpp there is a struct defined as:

`

    typedef struct {
        hSSurface d[100];
    } Revolved;
`
This was originally used in Lathe groups to temporarily collect 4 90 degree sections of a surface. I expanded it to 100 to support more sections in Helix groups, but having a fixed number in there is not so great.

Also, in one of my branches (draft) I wanted to pass a list of these to a function and so had to move the definition into surface.h which to me is also a bit undesirable.

A better container for these would be nice. Is the right answer just a list? I'm not fan of doing things like list of lists in C++ but even that is better than making this a fixed size. @rpavlik ?

rpavlik, 2019-08-22:
Well, if it needs the slightly-weird semantics of List<> (that is, calling clear on everything before removing it), then use List<> otherwise use std::vector directly.

whitequark, 2019-08-22:
We don't really have a policy on using `List` vs `std::vector` in existing code. I've used `std::vector` in parts that are self-contained and completely new (like new export functions), but I'd say if it's a part of the geometry code then `List` would be less annoying to work with.

But this is fairly minor and just my opinion.

phkahler, 2019-08-24:
std::vector worked out nicely because we know what length is needed when they're created.

phkahler, 2019-08-25:
This was done in PR #466 commits ced879c and cad7002

closed 2019-08-25 by phkahler