solvespace/solvespace #469: Changing width in "#def-draw-error" style has no effect

Symbian9, 2019-09-02:
### System information

SolveSpace version: `3.0~22e4011c`

Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

Changing any parameter of any style should work as expected.

### Actual behavior

Changing  width in `#def-draw-error` style has no effect.

[image]

### Additional information

- SLVS - https://gist.github.com/Symbian9/d38ce6749ba48c27a96fd6e9ac1ddbd3

whitequark, 2019-09-02:
What really should happen here is that the built-in styles shouldn't allow changing any items that don't actually affect the text. But this is fairly annoying to track down.

Symbian9, 2019-12-01:
> Changing width in `#def-draw-error` style has no effect.

Found that similar issue happen with `#def-selected` line style (only color changing has effect for this style), but it reuse width and stipple type from `#def-active-grp` instead.

whitequark, 2019-12-01:
That's intended. The bug here is that the built-in style lets you change this parameter.

Symbian9, 2019-12-01:
> The bug here is that the built-in style lets you change this parameter.

Sometime, changing built-in styles are really needed.

For example, if I want set "red" background, default built-in "red" highlighting for selected entities is mostly useless.

[image]

User should has possibility to tune default UI & styles according personal preferences

whitequark, 2019-12-01:
> Sometime, changing built-in styles are really needed.

You misunderstood. Being able to change e.g. selected entity color is useful and intended. The problem is that although the `#def-selected` style exists only to allow you to change selected entity color, it shows many more changeable parameters that do not affect anything. These parameters should not be shown.