solvespace/solvespace #470: Union of "Revolve" and "Extrude" groups failed

Symbian9, 2019-09-02:
### System information

SolveSpace version: `3.0~22e4011c`

Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

If path for «Extrude» group based on plane from «Revolve» group boolean union operation should be correct.

### Actual behavior

Boolean union of parent «Revolve» and child «Extrude» (based on plane face from «Revolve») groups failed.

[image]

### Additional information

- SLVS - https://gist.github.com/Symbian9/d38ce6749ba48c27a96fd6e9ac1ddbd3

whitequark, 2019-09-02:
Like the other NURBS backend bugs, don't expect this to be fixed.

closed 2019-09-02 by whitequark

ruevs, 2019-09-02:
The situation you are creating is in fact a "singularity" and is very hard for 3D software to handle. The handle and the bowl of the "spoon" "meet" only at the plane on the bowl that you are extruding the handle from. This plane is the problem - it is the only "common" entity between the two  solids - the handle and the bowl - they have no volume in common. Therefore when you do a union of these two solids it becomes a very tricky question whether the result is a single solid or not. It depends on whether the face "belongs" to the solids or not...

In terms of "sets":

* [bowl][handle] - the union is a solid
* [Bowl)[Handle] and [Bowl](Handle] - the union should be a solid but...
* [bowl)(handle] - the union is not a solid - there is an infinitely thin plane that does not belong to the spoon.

The way to avoid this situation in this case is to extrude a double sided "stub" from the plane and then extrude the handle from the same plane. In general when modeling you should avoid solids that meet at single planes/edges/points - all of these create "singularities" and based on how "good" the NURBS code is and how much effort was spent to handle such cases it may or may not work.

ruevs, 2019-09-02:
Here is an explanation from Autodesk much better than mine above :-)
https://blogs.autodesk.com/inventor/2019/02/13/seven-deadly-sins-3d-part-modeling-part-03-singularities/

ruevs, 2019-09-02:
https://blogs.autodesk.com/inventor/2019/02/06/7-deadly-sins-3d-part-modelling-part-02-near-coincidence/

phkahler, 2019-09-06:
I tried this. Not sure how you defined the workplane for the extrude sketch. I had to add a construction line perpendicular to the top edge. It is helpful to make the top straight edge out of two line segments to match the two that resulted from the revolve. Also not sure how you matched the diameter of the arc in the sketch to the revolved surface since there isn't an arc to set equal diameters (you can do it other ways).

I never really considered the special case where the revolve is 180 degrees and the two flat surfaces could become one.

Symbian9, 2019-09-07:
> Not sure how you

I will try record screencast with my workflow.

> I had to add

@phkahler, please, provide your SLVS file.

nccvector, 2025-05-13:
I am trying to create an intersection of pipes and I am having difficulty taking the union of two intersecting cylinders:

[image]

I do not want to force it to triangles because I want to keep surfaces.

I also tried it at an angle:

[image]

nccvector, 2025-05-13:
Okay after reading some threads I have figured out a hacky way to achieve the above drawing. The trick was to add a tiny offset <=0.1mm between the axes of two cylinders. I also believe that if the cylinder axes intersect exactly, then there is some sort of "singularity" that the system struggles with.
Now adding this tiny offset felt like a shame, but nevertheless it got the job done:

[image]

Tiny offset 0.1:

[image]