solvespace/solvespace #478: selection and unselection with mouse clicks.

madflower, 2019-09-10:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...) Head and the release version

Operating system: (e.g. Debian testing)
Fedora 30
### Expected behavior

What should have happened?
I should be able to unselect with a mouse click
### Actual behavior
It doesn't unselect properly. 
What actually happened?

### Additional information

This seems to work: 
`diff --git a/src/draw.cpp b/src/draw.cpp
index ee3966b..01f51f2 100644
--- a/src/draw.cpp
+++ b/src/draw.cpp
@@ -181,8 +181,12 @@ void GraphicsWindow::MakeSelected(hConstraint hc) {
 
 void GraphicsWindow::MakeSelected(Selection *stog) {
     if(stog->IsEmpty()) return;
-    if(IsSelected(stog)) return;
-
+  //  if(IsSelected(stog)) return;
+   if(IsSelected(stog))
+   {
+        MakeUnselected(stog, /*coincidentPointTrick=*/false);
+        return;
+   }
     if(stog->entity.v != 0 && SK.GetEntity(stog->entity)->IsFace()) {
         // In the interest of speed for the triangle drawing code,
         // only two faces may be selected at a time.
`

whitequark, 2019-09-10:
> I should be able to unselect with a mouse click

I don't consider this a bug. SolveSpace had the current selection behavior for many years, and changing it would only serve to confuse anyone who is used to it.

whitequark, 2019-09-10:
If you wanted to unselect something with a mouse click, you can Ctrl+click it instead. This was added in #106.

closed 2019-09-10 by whitequark