solvespace/solvespace #482: Transfer snap ownership to solvespace organization

ppd1990, 2019-09-11:
With https://github.com/solvespace/solvespace/pull/436 being almost ready for merging, I propose the following steps regarding the publishing of the snap:

- [ ] Create a ```solvespace ``` account on snapcraft.io
This gives the solvespace snap store listing an official touch and at some point in the future, it can be verified by the store and appear with a little badge or checkmark.
- [ ] I request to transfer ownership of the snap to the newly created account on the snapcraft forum: https://forum.snapcraft.io/c/store-requests
- [ ] If necessary, the snap's administration can be shared with other accounts. A shared secret is not necessary. The link for that would be https://dashboard.snapcraft.io/snaps/solvespace/collaboration/

whitequark, 2019-11-23:
@jwesthues Could you create an account on snapcraft.io for SolveSpace and add me as a collaborator?

jwesthues, 2019-11-23:
Cool, `solvespace` account created. Let me know when the transfer is complete, and I'll add you (and my own personal account, and whoever else we should have) as collaborators.

ppd1990, 2019-11-26:
The transfer request was filed here:
https://forum.snapcraft.io/t/please-transfer-solvespace-to-solvespace-account/14342

ppd1990, 2019-11-28:
@jwesthues Have you been contacted by someone from the Snap Store? I presume they'll require your OK first. You could probably also claim the `solvespace` name from your account. That'll get the ball rolling a little faster, I think.

jwesthues, 2019-11-28:
No contact yet. I'm traveling now, but will try claiming it if no action by my return Sunday.

jwesthues, 2019-12-01:
No contact yet, sent the claim today.

ppd1990, 2019-12-04:
Nothing yet. Will poke the store admins some more...

ppd1990, 2019-12-04:
Poking was successful. I've been contacted already and have confirmed the transfer request. @jwesthues You should you have mail from Canonical

ppd1990, 2019-12-04:
The transfer has been completed. Please note that I am still (by default) an active collaborator of the snap. So don't forget to remove me if necessary.

jwesthues, 2019-12-04:
I got the transfer, added @whitequark as a collaborator, left @ppd1990 just in case. Thanks, and let me know if any other action required.

whitequark, 2020-01-21:
@jwesthues It looks like I can't access the snap--can you recheck?

ppd1990, 2020-01-21:
@whitequark Your invite is still pending. You need to respond to the email

whitequark, 2020-01-21:
Thanks, that worked.

closed 2020-01-21 by whitequark

whitequark, 2020-01-23:
> The transfer has been completed. Please note that I am still (by default) an active collaborator of the snap. So don't forget to remove me if necessary.

I've removed you to minimize potential risk of compromise in the future. Thank you for your work on bringing this together!

ppd1990, 2020-01-23:
Thanks for reviewing. I'll PR some readme changes to guide users to the snap store soon.