solvespace/solvespace #483: Snap Travis integration

ppd1990, 2019-09-11:
As outlined in https://github.com/solvespace/solvespace/pull/436 we'd like  to have build on commit and also release to the edge channel.

## Using ```snapcraft remote-build```

The current plan involves using ```snapcraft remote-build``` to build for all possible architectures. This feature is still in beta @ https://github.com/snapcore/snapcraft/tree/remote-build and will receive further simplifications, so I propose waiting for that to be properly finished.

A sample implementation is here: https://github.com/ppd1990/solvespace-travis

## Using docker

ppd1990, 2019-10-30:
And here we go: https://forum.snapcraft.io/t/call-for-testing-snapcraft-3-9/13944

Snapcraft 3.9 is pre-released and includes remote-build support. Let's do it.

whitequark, 2019-11-23:
Sounds good. Let me know once it becomes stable.

ppd1990, 2019-11-23:
Certainly. We're going to need a solvespace launchpad account for remote builds though. I'll test with mine in the meantime

ppd1990, 2019-11-23:
Is there a particular reason for ```dist: trusty```? The Travis snap addon requires xenial or above and I'd rather use that if possible.

whitequark, 2019-11-23:
I don't recall if there was. It should be fine to upgrade to Xenial.

ppd1990, 2020-01-23:
This was implemented by: 
1. a60d4df
2. d5351c4
3. 2dcd81d

Snaps are being built and deployed correctly: https://travis-ci.org/solvespace/solvespace/jobs/640723069

I consider this completed.

whitequark, 2020-01-23:
Is this process supposed to upload 4 separate releases, each supporting 1 architecture? Seems suboptimal.

ppd1990, 2020-01-23:
The revision is a strictly increasing integer per project and applies to all uploaded snaps. So yes, it is indeed meant to be that way.

See `qelectrotech`'s revision matrix for example, which uses Canonical's "zero-config" https://snapcraft.io/build service:

[image]

whitequark, 2020-01-23:
We're all done here!

closed 2020-01-23 by whitequark