solvespace/solvespace #488: Feature request: Improve percision of helix extrusion

kalle, 2019-09-20:
To make it easier to crate threads that should match some predefined pitch, I suggest that the pitch of the helix extrusion can be entered in a text box for that group. The height of the helix can be set in the diagram. The current solution with free-hand twisting makes it hard to get a desired pitch.

phkahler, 2019-09-25:
This should be a "pitch" parameter in the text window for the group. That parameter would be used to constrain the helix angle and the length along the axis.  distance = (angle/2pi) * pitch.

From an implementation point of view, I'm not sure what to do when the user doesn't specify a pitch value, or how to dynamically create or remove the constraint equation from the group when changed in the text window. For helix, the angle/2 is in param[3] and the distance in param[7].

whitequark, 2019-09-26:
@phkahler I know how to do this. I'll try and implement it once I get around to reviews/merges (soon).

phkahler, 2020-05-16:
@whitequark  I'm really interested in how to do this. I know how to create an internal constraint when a group is created, but this one wants to be created afterword (and maybe deleted) when the user may have created others manually.

closed 2021-07-31 by phkahler