solvespace/solvespace #490: Crash when clicking "other angle" when there is no angle to act on.

rpavlik, 2019-09-21:
### System information

SolveSpace version: 5d78f993cea3749b6644e0a2f9ddf89c9754e377 - Windows artifact from AppVeyor

Operating system: Windows 10

### Expected behavior

When I accidentally clicked "other complementary angle" instead of "constraint angle" (with 2 line segments selected) I assume it was supposed to tell me I goofed with a message box.

### Actual behavior

It crashed.

### Additional information

I hadn't saved yet, but the thing I was making wasn't that complicated (no curves) I doubt it will help, but the first group in the project was this file "linked" as a master document of sorts.
[keepout-volume.zip](https://github.com/solvespace/solvespace/files/3639072/keepout-volume.zip)

whitequark, 2019-11-23:
Can't reproduce here. Could you grab a backtrace?

rpavlik, 2020-03-07:
ah ha. I can't reproduce in my local build (with vs2019) but this, as well as other things that should pop up a dialog (drawing  a rectangle without a workplane, etc) all crash the AppVeyor build. Looks like there's something wrong with that build which is causing crashes instead of popups.

whitequark, 2020-03-08:
There's debug information uploaded from AppVeyor, do you think you can use that in MSVC to grab a backtrace?

rpavlik, 2020-03-08:
ah, yeah probably, will give it a shot

rpavlik, 2020-03-10:
"Unhandled exception at 0x014C0713 in solvespace.exe: An invalid parameter was passed to a function that considers invalid parameters fatal. occurred"

```
>	solvespace.exe!_invoke_watson(const wchar_t * pszExpression, const wchar_t * pszFunction, const wchar_t * pszFile, unsigned int nLine, unsigned int pReserved) Line 132	C++
 	solvespace.exe!_invalid_parameter(const wchar_t * pszExpression, const wchar_t * pszFunction, const wchar_t * pszFile, unsigned int nLine, unsigned int pReserved) Line 86	C++
 	solvespace.exe!std::_String_const_iterator<std::_String_val<std::_Simple_types<char>>>::operator*() Line 80	C++
 	solvespace.exe!std::_String_iterator<std::_String_val<std::_Simple_types<char>>>::operator*() Line 336	C++
 	solvespace.exe!MessageBox(const char * fmt, char * va, bool error, std::function<void __cdecl(void)> onDismiss) Line 142	C++
 	solvespace.exe!SolveSpace::Error(const char * fmt, ...) Line 183	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::MouseLeftDown(double mx, double my, bool shiftDown, bool ctrlDown) Line 995	C++
 	solvespace.exe!SolveSpace::GraphicsWindow::MouseEvent(SolveSpace::Platform::MouseEvent event) Line 898	C++
 	solvespace.exe!std::_Pmf_wrap<bool (__thiscall SolveSpace::GraphicsWindow::*)(SolveSpace::Platform::MouseEvent),bool,SolveSpace::GraphicsWindow,SolveSpace::Platform::MouseEvent>::operator()(SolveSpace::GraphicsWindow * _Pfnobj, SolveSpace::Platform::MouseEvent <_Fargs_0>) Line 1230	C++
 	solvespace.exe!std::_Bind<1,bool,std::_Pmf_wrap<bool (__thiscall SolveSpace::GraphicsWindow::*)(SolveSpace::Platform::MouseEvent),bool,SolveSpace::GraphicsWindow,SolveSpace::Platform::MouseEvent>,SolveSpace::GraphicsWindow * const,std::_Ph<1> &>::_Do_call<SolveSpace::Platform::MouseEvent,0,1>(std::tuple<SolveSpace::Platform::MouseEvent &> _Myfargs, std::_Arg_idx<0,1> __formal) Line 1150	C++
 	solvespace.exe!std::_Bind<1,bool,std::_Pmf_wrap<bool (__thiscall SolveSpace::GraphicsWindow::*)(SolveSpace::Platform::MouseEvent),bool,SolveSpace::GraphicsWindow,SolveSpace::Platform::MouseEvent>,SolveSpace::GraphicsWindow * const,std::_Ph<1> &>::operator()<SolveSpace::Platform::MouseEvent>(SolveSpace::Platform::MouseEvent && <_Fargs_0>) Line 1138	C++
 	solvespace.exe!std::_Callable_obj<std::_Bind<1,bool,std::_Pmf_wrap<bool (__thiscall SolveSpace::GraphicsWindow::*)(SolveSpace::Platform::MouseEvent),bool,SolveSpace::GraphicsWindow,SolveSpace::Platform::MouseEvent>,SolveSpace::GraphicsWindow * const,std::_Ph<1> &>,0>::_ApplyX<bool,SolveSpace::Platform::MouseEvent>(SolveSpace::Platform::MouseEvent && <_Args_0>) Line 284	C++
 	solvespace.exe!std::_Func_impl<std::_Callable_obj<std::_Bind<1,bool,std::_Pmf_wrap<bool (__thiscall SolveSpace::GraphicsWindow::*)(SolveSpace::Platform::MouseEvent),bool,SolveSpace::GraphicsWindow,SolveSpace::Platform::MouseEvent>,SolveSpace::GraphicsWindow * const,std::_Ph<1> &>,0>,std::allocator<std::_Func_class<bool,SolveSpace::Platform::MouseEvent>>,bool,SolveSpace::Platform::MouseEvent>::_Do_call(SolveSpace::Platform::MouseEvent && <_Args_0>) Line 229	C++
 	solvespace.exe!std::_Func_class<bool,SolveSpace::Platform::MouseEvent>::operator()(SolveSpace::Platform::MouseEvent <_Args_0>) Line 315	C++
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WndProc(HWND__ * h, unsigned int msg, unsigned int wParam, long lParam) Line 951	C++
```

rpavlik, 2020-03-10:
Specifically, the MessageBox issue location is:

```
   142:     while(isspace(*it)) it++;
00DCEC69  movsx       ecx,byte ptr [eax]  
00DCEC6C  push        ecx  
00DCEC6D  call        isspace (014C5FD0h)  
00DCEC72  add         esp,4  
00DCEC75  test        eax,eax  
00DCEC77  je          MessageBox+29Ah (0DCEC9Ah)  
00DCEC79  push        0  
00DCEC7B  lea         edx,[ebp-144h]  
00DCEC81  push        edx  
00DCEC82  lea         ecx,[it]  
00DCEC88  call        std::_String_iterator<std::_String_val<std::_Simple_types<char> > >::operator++ (0DD1300h)  
00DCEC8D  lea         ecx,[ebp-144h]  
00DCEC93  call        std::_Vector_iterator<std::_Vecto
```

Wonder if this is the old "std::string(nullptr) goes boom" or something like it...

closed 2020-03-12 by whitequark