solvespace/solvespace #494: crash when trying to use "terminus" font

bmorel, 2019-10-20:
SolveSpace version: from git, 5d78f993cea3749b6644e0a2f9ddf89c9754e377

Operating system: Debian Buster

### Expected behavior

Showing some text rendered with Terminus or display an error because of failure.

### Actual behavior

Crash. Here is the log:

```
Generate::ALL took 184 ms
Missing (absent) translation for 'click to place bottom right of text'
Generate::DIRTY took 40 ms
Generate::DIRTY took 40 ms
freetype: cannot set character size: invalid pixel size
File /home/foobar/contrib/solvespace/src/solvespace/src/ttf.cpp, line 404, function AspectRatio:
Assertion failed: fontFace != NULL.
Message: Expected font face to be loaded.
zsh: abort      solvespace
```

The message about "cannot set character size" is thrown by ttf.cpp:234, content of sizeRequest is:
{type = FT_SIZE_REQUEST_TYPE_REAL_DIM, width = 65536, height = 65536, horiResolution = 128, vertResolution = 128}

Error value is not used or sent to caller on ttf.cpp:102, should be checked and probably return NULL. With other things, I guess.

whitequark, 2019-10-20:
Would you like to take a stab writing a PR?

bmorel, 2019-10-20:
I don't know what should be the solvespace behavior, I only did my best to make it easier to people with real understanding of it's codebase to fix that. If I knew enough the code, I'd probably try to implement support for psf format (fixed size, used in linux terminals, should be way faster to render since no real curves).

whitequark, 2019-11-23:
Fixed in 14e095c9.

closed 2019-11-23 by whitequark