solvespace/solvespace #495: crash when trying to opening a file

bmorel, 2019-10-20:
### System information

SolveSpace version: 5d78f993cea3749b6644e0a2f9ddf89c9754e377
Operating system: Debian buster

### Expected behavior

File should be open.

### Actual behavior

Crash with that log:

```
File /home/foobar/contrib/solvespace/src/solvespace/src/platform/platform.cpp, line 400, function OpenFile:
Assertion failed: filename.raw.length() == strlen(filename.raw.c_str()).
Message: Unexpected null byte in middle of a path.
zsh: abort      solvespace
```

### Additional information

This crash seems to be semi-random, can't find a way to reproduce it. Seems to happen when using "Recently open" list. I wish I could do more.

whitequark, 2019-10-20:
You could try running it under gdb and see what `filename.raw` looks like

bmorel, 2019-10-20:
Sure, but for that I'd need to run it under gdb everytime since I can't find how to reproduce. Or maybe I could patch the assert to dump hexa code of both version of filename... seems less annoying to do that. I'll try to fetch more info.

mePy2, 2020-06-02:
I have the same problem.

macOS Mojave

Solvespace opens other files correctly. Not this one though.

whitequark, 2020-06-02:
Does it always reproduce with that file? If so, can you upload it?

mePy2, 2020-06-02:
Yeah... I mean, I created it and modified it with Solvespace. But now it will crash the app totally!
Since I have Wine, I downloaded the exe file and it does work and can open the file without issue.

Anyway, while typing this, I made a quick test with the Windows exec and I found out the problem was with the `force NURBS surfaces to triangle mesh` option.
I simply opened the file with WINE, deselected the option, saved and it now works with Solvespace for macOS too.

whitequark, 2020-06-02:
Does it crash with the "Unexpected null byte in middle of a path." assertion on macOS for you?

mePy2, 2020-06-02:
I got mad. And very unhappy, because I love Solvespace.
Also, while trying a way to open the file, I tried to download the `.dmg` file from here and I found out that it still does not work “out of the box”. (I wanted to see if the issue happened with a fresh Solvespace app package too.)

mePy2, 2020-06-02:
The app does not quit itself. The rolling circle appears and the app does not respond. Everything freeze.

mePy2, 2020-06-02:
Anyway now the problem happened again but just by _selecting_ the `force NURBS surfaces to triangle mesh` box.

The only way to stop it is to force close it.

whitequark, 2020-06-02:
If you don't upload the file that reproduces the bug (or otherwise provide a series of steps that leads to the crash) there is nothing I can do to fix it.

mePy2, 2020-06-02:
Yeah sure, may I send it to you privately?

whitequark, 2020-06-02:
Sure. whitequark@whitequark.org.

whitequark, 2020-06-02:
Okay, I got your file. Your crash is not related to this issue, and does not reproduce on master, so I suspect it has been fixed since 2.3. Annoyingly right now there's no easy way to get a recent `.dmg` but #618 should fix that.

mePy2, 2020-06-02:
Ok... so you suggest to compile a recent version?

whitequark, 2020-06-02:
Yep. I don't currently have a reasonably functional way to develop for macOS so I can't easily build it for you.

mePy2, 2020-06-02:
Hi,

I built Solvespace for macOS. The app package is in the bin folder. There is also the `.dmg` file. You could put that one in the _Releases_. Also, the dmg opens and Solvespace works without problems; “out-of-the-box”.

There is an issue though, the screen flashes continuously. flickers while moving the mouse cursor.

whitequark, 2020-06-02:
> If you want I can upload the dmg to be put in _GitHub Releases_.

The files in GitHub Releases are populated exclusively by automated build scripts so that they are (in principle) reproducible by anyone who wants to ensure they're legitimate, and there is no chance that a developer uploads a binary infected by malware.

> There is an issue though, the screen flashes continuously. flickers while moving the mouse cursor.

That's unfortunate. Someone with both Apple hardware and good understanding of macOS graphics would have to fix that.

mePy2, 2020-06-02:
Oh cool. I didn’t know that but I totally agree with that.

About the Mac issue, yeah. That’s very unfortunate. The old package does not do that though.

PS:
I like the old UI/color the most.

mePy2, 2020-06-02:
If I will care enough – lol –, I would like to port Solvespace to Mac OS X Tiger PowerPC.

whitequark, 2020-06-02:
> The old package does not do that though.

Sure. The renderer was completely rewritten in the meantime for a variety of reasons so it's not trivial to determine what change causes the flickering. Moreover, the new renderer works just fine on any version of macOS I have access to, so this issue also might be caused by a change in macOS.

> I like the old UI/color the most.

The UI colors were changed to have higher contrast to aid visually impaired and colorblind users.

mePy2, 2020-06-02:
Oh...

ruevs, 2020-11-26:
@bmorel @mePy2 does the current edge macOS build crash as you described in this issue?
What about the "flickering"?
https://github.com/solvespace/solvespace/releases/tag/edge

mePy2, 2020-12-04:
> @bmorel @mePy2 does the current edge macOS build crash as you described in this issue?
> What about the "flickering"?
> https://github.com/solvespace/solvespace/releases/tag/edge

Hi @ruevs I’m sorry for the late reply. I did’t forgot about it though.
I am going to try it as soon as possible. Thank you very much.

mePy2, 2020-12-11:
@ruevs It works. Thank you very much.
Would you mind telling me what the problem was?
Cheers

ruevs, 2020-12-11:
I don't know. Too many things have changed since 2.3, I don't have a Mac and the file you sent to Whitequark ;-)

I asked you to try since we now have an always fresh macOS build.

tomsci, 2021-04-04:
I hit this myself (on OSX, master build from yesterday) and what I think is happening is that in `GraphicsWindow::PopulateRecentFiles()`, the call to `SS.OkayToStartNewFile()` can potentially update `recentFiles`, which invalidates the `path` reference - which will only break things if the `path` actually moves in the vector, which explains why it doesn't reproduce all the time.

My steps to reproduce are:
Open File1. Make some changes to it
Open File2 from the Open Recent menu. When asked whether to save changes to File1, click Save.
Do this a couple of times (maybe the number of elements in the recents list is a factor, or which element you open) and you'll hit that assert eventually.

I'll open a PR shortly.

closed 2021-04-04 by phkahler