solvespace/solvespace #496: cmake includes absolute filenames in the binary

bmorel, 2019-10-20:
I know it's not related to solvespace itself, but I find it annoying and I know there is a [workaround](https://stackoverflow.com/questions/9607155/make-gcc-put-relative-filenames-in-debug-information). Dunno if that would be ok for solvespace, though, since it probably does not work for Visual Studio.

I use it at work with gcc and clang, haven't seen any problem for now, only benefits (like, allows to debug softwares built by a colleague without needing tricks because of user name is in the path). Never tried it with an IDE, nor with Visual Studio: i3 + zsh + vim + cgdb are as powerful as many IDE I know (and crash less).

I'd be ok to take some time writing a patch if such "feature" would be accepted.

whitequark, 2019-10-20:
There are two solutions behind your link. I think using `-fdebug-prefix-map` in the case where GCC/Clang are used is perfectly fine. I think using a shell script to rewrite options is too fragile.

bmorel, 2019-10-20:
I think `-fdebug-prefix-map` didn't worked for me, unlike the script. I'll verify that tomorrow on projects I really know how they work.

whitequark, 2019-10-20:
One issue is that e.g. Debian packages use their own `-fdebug-prefix-map`, so this feature should be possible to turn off.

whitequark, 2019-11-23:
Fixed in 9e7681f2.

closed 2019-11-23 by whitequark