solvespace/solvespace #498: Crash: Opening the application after downloading.

NovaBro, 2019-10-28:
### System information

SolveSpace version: 2.3
*cant use the trick using the "use Help → About..." since it wont even open

Operating system: MacOS

Expected:
I would expect for it to launch after downloading and opening the 2.3 solvespace.dmg file.

What Happened:
It did not open and immediately crashed.

https://pastebin.com/3n5qgZB9

whitequark, 2019-10-28:
See the workaround here: https://github.com/solvespace/solvespace/issues/398

NovaBro, 2019-11-11:
I followed their instructions of deleting the files highlighted in blue, but I did not understand how to complete the first step.  
"brew install cmake libpng freetype
(see after the Installation "section" in "Building" from the the README.md)
Maybe it works without that"
Regardless, after deleting the files, I tried again but it did not work.

LdBeth, 2019-11-16:
https://brew.sh

this site gives your info on what is brew, where to get it, and how to use it.

For me in addition to deleting the dylib files in the bundle, to get the app work requires using `install_name_tool` to edit dynamic shared library name linked to the app.

```
>> 0:~/Downloads $ cd solvespace.app/Contents/MacOS/                                                       << 23:31 <]
>> 0:~/D/s/C/MacOS $ ls                                                                                    << 23:33 <]
libfreetype.dylib libpng.dylib      libz.dylib        solvespace
>> 0:~/D/s/C/MacOS $ rm  *.dylib                                                                           << 23:33 <]
override r--r--r--  ldbeth/staff for libpng.dylib? y
>> 0:~/D/s/C/MacOS $ ls                                                                                    << 23:33 <]
solvespace
>> 0:~/D/s/C/MacOS $ ./solvespace                                                                          << 23:33 <]
dyld: Library not loaded: @executable_path/libpng.dylib
  Referenced from: /Users/ldbeth/Downloads/solvespace.app/Contents/MacOS/./solvespace
  Reason: image not found
fish: './solvespace' terminated by signal SIGABRT (Abort)
>> 134:~/D/s/C/MacOS $ otool -L solvespace                                                                 << 23:33 <]
solvespace:
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	@executable_path/libpng.dylib (compatibility version 42.0.0, current version 42.0.0)
	@executable_path/libz.dylib (compatibility version 1.0.0, current version 1.2.5)
	@executable_path/libfreetype.dylib (compatibility version 19.0.0, current version 19.3.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1404.46.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices (compatibility version 1.0.0, current version 48.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1258.1.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1258.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
>> 0:~/D/s/C/MacOS $ ./solvespace                                                                                                                         << 23:34 <]
dyld: Library not loaded: @executable_path/libpng.dylib
  Referenced from: /Users/ldbeth/Downloads/solvespace.app/Contents/MacOS/./solvespace
  Reason: image not found
fish: './solvespace' terminated by signal SIGABRT (Abort)
>> 134:~/D/s/C/MacOS $ install_name_tool -change @executable_path/libpng.dylib /opt/pkg/lib/libpng16.dylib solvespace                                     << 23:36 <]
>> 0:~/D/s/C/MacOS $ ./solvespace                                                                                                                         << 23:37 <]
dyld: Library not loaded: @executable_path/libfreetype.dylib
  Referenced from: /Users/ldbeth/Downloads/solvespace.app/Contents/MacOS/./solvespace
  Reason: image not found
fish: './solvespace' terminated by signal SIGABRT (Abort)
>> 134:~/D/s/C/MacOS $ install_name_tool -change @executable_path/libfreetype.dylib /opt/pkg/lib/libfreetype.dylib                                        << 23:37 <]
Usage: /Library/Developer/CommandLineTools/usr/bin/install_name_tool [-change old new] ... [-rpath old new] ... [-add_rpath new] ... [-delete_rpath old] ... [-id name] input
>> 1:~/D/s/C/MacOS $ install_name_tool -change @executable_path/libfreetype.dylib /opt/pkg/lib/libfreetype.dylib solvespace                               << 23:37 <]
>> 0:~/D/s/C/MacOS $ ./solvespace                                                                                                                         << 23:37 <]
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
Vector::WithMagnitude(2.6) of zero vector!
>> 0:~/D/s/C/MacOS $ otool -L solvespace                                                                                                                  << 23:38 <]
solvespace:
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/opt/pkg/lib/libpng16.dylib (compatibility version 42.0.0, current version 42.0.0)
	@executable_path/libz.dylib (compatibility version 1.0.0, current version 1.2.5)
	/opt/pkg/lib/libfreetype.dylib (compatibility version 19.0.0, current version 19.3.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1404.46.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
[... cut ...]
>> 0:~/D/s/C/MacOS $ install_name_tool -change @executable_path/libz.dylib /opt/pkg/lib/libz.dylib                                                        << 23:38 <]
Usage: /Library/Developer/CommandLineTools/usr/bin/install_name_tool [-change old new] ... [-rpath old new] ... [-add_rpath new] ... [-delete_rpath old] ... [-id name] input
>> 1:~/D/s/C/MacOS $ install_name_tool -change @executable_path/libz.dylib /opt/pkg/lib/libz.dylib solvespace                                             << 23:39 <]
```

NovaBro, 2019-11-20:
I was able to download HomeBrew and follow the original instructions. Thank you all for your help!

closed 2019-11-20 by NovaBro