solvespace/solvespace #499: CLI option "--view" has no effect on output thumbnail

Symbian9, 2019-11-02:
### System information

- **SolveSpace version:** `3.0~5d78f993` (`solvespace-cli`)
- **Operating system:** Debian 9.x stretch

### Expected behavior

Changing `--view` option variable to `top`/`bottom`/`left`/`right`/`front`/`back`/`isometric` should change view for rendering thumbnail.

### Actual behavior

I tried to generate thumbnail with `--view isometric` direction:

```
$ solvespace-cli thumbnail --output ./%-render.png --size 1280x800 --view isometric ./dt830b-case-top.slvs
```

But resulted image rendered in default *SolveSpace* view, e.g.:

```
projection onto screen:
  right  ( 1.000, 0.000, 0.000 )
  up     ( 0.000, 1.000, 0.000 )
  out    ( 0.000, 0.000, 1.000 )
```

Tried different `--view` variables, but result always is same.

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

Symbian9, 2019-11-02:
As I understand, `--view isometric` (nearest isometric) projection should be something like:

```
projection onto screen:
  right  ( 0.707, 0.000, -0.707)
  up     (-0.408, 0.816,-0.408)
  out    ( 0.577, 0.577, 0.577)
```

Is it possible add option *«save as default isometric view»* into *«Properties Browser > view > projection onto screen:»* (+ «load factory default» for reset) to define save actual view as default "isometric" projection?

phkahler, 2020-09-18:
I wonder if this is related to #567.

phkahler, 2020-09-19:
I merged PR #706 but I don't think this issue is ready to close. It's not clear to me that the options top,bottom, etc... are producing the correct views and they are not always centred well. Isometric seems to work though.

What is the default view in SolveSpace? Is that considered "front"? Or is it "top"? or something else?

phkahler, 2020-09-19:
The camera offset in GraphicsWindow::ZoomToFit is still not centering the view correctly.

closed 2020-09-19 by phkahler