solvespace/solvespace #501: Crash on Startup

asantoni, 2019-11-22:
### System information

SolveSpace version: 3.0 git revision 5d78f993cea3749b6644e0a2f9ddf89c9754e377

Operating system: Windows 10

### Expected behavior 

No crash...

### Actual behavior

[image]

### Additional information

This was working yesterday for me, but mysteriously stopped working today. Sorry for the image paste of the backtrace, but I hope it's still useful. I'm not sure I have the resolve to track down this crash further myself...

asantoni, 2019-11-22:
Sorry, that backtrace was cut off... Here's the full backtrace:

[image]

whitequark, 2019-11-22:
Does this patch help?
```diff
diff --git a/src/render/gl3shader.cpp b/src/render/gl3shader.cpp
index 8ad45c7a..542eb97f 100644
--- a/src/render/gl3shader.cpp
+++ b/src/render/gl3shader.cpp
@@ -390,7 +390,7 @@ GLuint Generate(const std::vector<double> &pattern) {
     glGetIntegerv(GL_MAX_TEXTURE_SIZE, &size);
     RgbaColor *textureData = new RgbaColor[size];
 
-    int mipCount = (int)log2(size) + 1;
+    int mipCount = (int)log2(size);
     for(int mip = 0; mip < mipCount; mip++) {
         int dashI = 0;
         double dashT = 0.0;
```

asantoni, 2019-11-22:
Yep, that fixed it! Thanks for the quick fix!

asantoni, 2019-11-22:
Whoops, I should have left closing this ticket up to you.

whitequark, 2019-11-22:
> This was working yesterday for me, but mysteriously stopped working today.

Did you perhaps rebuild the application with a debug C++ runtime?

whitequark, 2019-11-22:
Fixed in 91ac6665.

closed 2019-11-22 by whitequark