solvespace/solvespace #511: Constraint for group A gets misapplied to the last Linked group during assembly

Zwahili, 2019-11-25:
### System information

SolveSpace version: 2.3~7c1ca460

Operating system: W7 Pro N SP1, x64

### Expected behavior

A "Constrain point on line..." operation applies the constraint to the selected group (done by clicking a point of the group and then one of the g001 ref planes). 

### Actual behavior

The constraint was placed on another group entirely; the one that was added last to the .slvs file.
This causes an "unsolved constraint" error.

### Additional information

[solvespace_constraint_missapl.zip](https://github.com/solvespace/solvespace/files/3888207/solvespace_constraint_missapl.zip)

In the attached project:
Assembly: assy2.slvs
Part 1: p1.slvs
Part 2: **not used here...**
Part 3: p3.slvs
Part 4: p4.slvs

assy2.slvs (and the error) was created by:
1: Adding (Link/Assemble...) p1, rotating it and adding its constraints
2: Adding p3, rotating it a bit
3: Adding p4, leaving it where it spawns
4: trying to add a "Constrain point on line..." between the marked point ("1: Select me!") on p3 and the #XY plane (from g001 ref planes)
5: The error (SOLVE FAILED! unsolvable constraints for c003-pt-in-plane) is produced
6: Pressing home in the error dialogue shows that there is only an error with p4 constraints
7: Entering p4 shows that the constraint that is causing the error is c003-pt-in-plane

The reason why p2 is not used is that I got the error before I could insert it (was trying to recreate the fundamentals of a more advanced assembly for this bug report).
However, if p2 is inserted and the steps above are followed again (while first having deleted the incorrect constraint), it is now p2 that gets the constraint instead of p4.

whitequark, 2019-11-26:
Thank you for the report. Can you reproduce this on master? You can download a nightly build [here](https://ci.appveyor.com/api/buildjobs/m4mik0q09ecyni7x/artifacts/build%2Fbin%2FDebug%2Fsolvespace.exe).

Zwahili, 2019-11-26:
> 
> 
> Thank you for the report. Can you reproduce this on master? You can download a nightly build [here](https://ci.appveyor.com/api/buildjobs/m4mik0q09ecyni7x/artifacts/build%2Fbin%2FDebug%2Fsolvespace.exe).

Could not even open the assembly in 3.0~a60d4df1 (eternally busy, not responding) so I had to build it again by linking p1/p3/p4. Same problem occurs, in the same way.

whitequark, 2019-11-26:
Interesting. Opening the assembly here causes an internal error:
```
File ./src/dsc.h, line 429, function FindById:
Assertion failed: t != NULL.
Message: Cannot find handle.
Aborted
```