solvespace/solvespace #518: "def-constraint" text characters has different width and color in exported PDF

Symbian9, 2019-12-01:
### System information

- SolveSpace version: `3.0~690f87c`
- Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

In exported PDF all characters of dimensions and comments text should follow "def-constraint" style settings.

### Actual behavior

This is what I get by default 

[image]

[image]

And this is what I get if change line stroke width from `1.0` (factory defaults style) to `2.0`

[image]

[image]

### Additional information

- SLVS - https://gist.github.com/Symbian9/644aff61c9dda1875c041c82a7cd54d4

My SolveSpace configuration preferences:

[image]

whitequark, 2019-12-01:
Good catch. This is caused by Z-fighting interacting with hidden line removal, I think.

Symbian9, 2020-05-14:
> This is caused by Z-fighting interacting with hidden line removal, I think.

Any code review on this issue? All dimensions labels in exported PDFs looks ugly due to this bug.

whitequark, 2020-05-14:
> Any code review on this issue?

No. And if you keep complaining I will ban you. Dealing with you has been the single most unpleasant experience I've ever had when working on open source, and I have considered giving up on SolveSpace entirely a few times just to never interact with you again. You have no concept of boundaries, no understanding of the needs of others, and no ability to correct your behavior when asked to.

robnee, 2021-02-28:
I think this had to do with how the code was applying styles to sets of closed loop paths.  Notice all of the properly styled constraint text above are loops.  The styling show be applied at the loop level not the set level.  See #946

phkahler, 2021-02-28:
I was able to reproduce the original problem but only the distance constraint showed it, my comment did not. Then I applied the patch from PR #946 and I got consistent colored constraints - all purple in the PDF. @Symbian9  is that the correct? It seems like one should have to redefine colors if they want them different in the .pdf, but the solid model is white and the edges are black, so I really don't know what to expect. It's definitely better to have the constraints all the same color!

Symbian9, 2021-02-28:
>  `@Symbian9` is that the correct?

@phkahler, I may not say nothing until this Pull Request would be merged into `master` and then packaged to DEB by @AlexanderP for Debian (as I can't build *SolveSpace* myself).

> It's definitely better to have the constraints all the same color!

This was expected by design (and that is the topic of this issue thread).

But I guess export of all, even custom styles, should be according each style properties. So, if user has few constraints with custom style (different in color & line drawing dashing, etc.) it should be exported according each constraint object style applied to it.

So, export all constraints in same style by ignoring applied custom styles would be just *another issue*.

phkahler, 2021-02-28:
@Symbian9  I merged the PR, so next time a package pops out that you can use please give it a try. Thanks.

robnee, 2021-03-01:
The export code does attempt to preserve all of the styles including custom styles.  There seemed to be an assumption that the way SS grouped sets of closed paths would result in sets of homogeneously styled paths.  It simply applied the styling of the first path to all of the paths in the set.  The sets are made op of paths of mixed styling and resulted in some paths incorrectly styled.

ruevs, 2021-03-01:
> I was able to reproduce the original problem but only the distance constraint showed it, my comment did not.

I am able to reproduce the problem for the comment and the constraint.

>Then I applied the patch from PR #946 and I got consistent colored constraints

Same here. And consistent comment as well.

In my opinion this is fixed in master 60dca4cb79

closed 2021-03-02 by ruevs