solvespace/solvespace #519: Impossible to compile under Linux LTS distributions

Symbian9, 2019-12-01:
### System information

- SolveSpace version: 3.0~690f87c
- Operating system: *Ubuntu 18.04 Bionic* ; *Debian 9.x stretch* (x86_64)

### Expected behavior

SolveSpace should keep support to build (at least) on Linux LTS distros

### Actual behavior

SolveSpace could not be compiled under Linux LTS distros, as `master` actually require *CMake* `3.11` dependency.

FTR, As non-trivial solution, user should try to compile *CMake* `3.11` manually before build *SolveSpace* `3.x`.

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2019-12-01:
What is the exact error you are getting from CMake?

Symbian9, 2019-12-01:
Think, @AlexanderP could provide more error details

AlexanderP, 2019-12-01:
Hi.
Ubuntu bionic
Cmake 3.10.2
```
dh_auto_configure -- -DCMAKE_LIBRARY_PATH=x86_64-linux-gnu -DCMAKE_BUILD_TYPE=Release -DENABLE_TESTS=OFF
	cd obj-x86_64-linux-gnu && cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON -DCMAKE_LIBRARY_PATH=x86_64-linux-gnu -DCMAKE_BUILD_TYPE=Release -DENABLE_TESTS=OFF
CMake Error at CMakeLists.txt:11 (cmake_policy):
  An attempt was made to set the policy version of CMake to "3.11" which is
  greater than this version of CMake.  This is not allowed because the
  greater version may have new policies not known to this CMake.  You may
  need a newer CMake version to build this project.

-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_FILE_PREFIX_MAP
-- Performing Test HAS_FILE_PREFIX_MAP - Failed
-- Using in-tree libdxfrw
-- Using in-tree flatbuffers
-- Looking for strtof_l
-- Looking for strtof_l - found
-- Using in-tree q3d
-- Looking for backtrace
-- Looking for backtrace - found
-- backtrace facility detected in default set of libraries
-- Found Backtrace: /usr/include  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.6.34") 
-- Found Freetype: /usr/lib/x86_64-linux-gnu/libfreetype.so (found version "2.8.1") 
-- Checking for module 'cairo'
--   Found cairo, version 1.15.10
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found SPACEWARE: /usr/lib/libspnav.so  
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.12.6
-- Checking for module 'json-c'
--   Found json-c, version 0.12.1
-- Checking for modules 'gtkmm-3.0>=3.18;pangomm-1.4;x11'
--   Found gtkmm-3.0, version 3.22.2
--   Found pangomm-1.4, version 2.40.1
--   Found x11, version 1.6.4
-- Configuring incomplete, errors occurred!
See also "/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
See also "/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/CMakeFiles/CMakeError.log".
	cd obj-x86_64-linux-gnu && tail -v -n \+0 CMakeCache.txt`
```
https://launchpadlibrarian.net/453851114/buildlog_ubuntu-bionic-amd64.solvespace_3.0.0~git1446-0472b9e-1ppa1~bionic1_BUILDING.txt.gz

whitequark, 2019-12-01:
Ok, the CMake behavior around `cmake_policy(VERSION)` doesn't make a lot of sense. Does this patch help?
```patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9bfd04d0..9bede12c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -7,8 +7,10 @@ if(CMAKE_SOURCE_DIR STREQUAL CMAKE_BINARY_DIR)
             "    mkdir build && cd build && cmake ..")
 endif()
 
-cmake_minimum_required(VERSION 3.2.0 FATAL_ERROR)
-cmake_policy(VERSION 3.11)
+cmake_minimum_required(VERSION 3.10.2 FATAL_ERROR)
+if(NOT CMAKE_VERSION VERSION_LESS 3.11.0)
+    cmake_policy(VERSION 3.11.0)
+endif()
 set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}
     "${CMAKE_SOURCE_DIR}/cmake/")
 set(CMAKE_CXX_STANDARD 11)
```

Symbian9, 2019-12-01:
> `+cmake_minimum_required(VERSION 3.10.2 FATAL_ERROR)`

FTR, Debian 9.x stretch has only CMake `3.7.2`

- https://packages.debian.org/stretch/cmake

AlexanderP, 2019-12-02:
> Ok, the CMake behavior around `cmake_policy(VERSION)` doesn't make a lot of sense. Does this patch help?
> 
> ```diff
> diff --git a/CMakeLists.txt b/CMakeLists.txt
> index 9bfd04d0..9bede12c 100644
> --- a/CMakeLists.txt
> +++ b/CMakeLists.txt
> @@ -7,8 +7,10 @@ if(CMAKE_SOURCE_DIR STREQUAL CMAKE_BINARY_DIR)
>              "    mkdir build && cd build && cmake ..")
>  endif()
>  
> -cmake_minimum_required(VERSION 3.2.0 FATAL_ERROR)
> -cmake_policy(VERSION 3.11)
> +cmake_minimum_required(VERSION 3.10.2 FATAL_ERROR)
> +if(NOT CMAKE_VERSION VERSION_LESS 3.11.0)
> +    cmake_policy(VERSION 3.11.0)
> +endif()
>  set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}
>      "${CMAKE_SOURCE_DIR}/cmake/")
>  set(CMAKE_CXX_STANDARD 11)
> ```

Ubuntu Bionic. Build was successful

Symbian9, 2019-12-02:
> Ubuntu Bionic. Build was successful

What about Debian stretch?

AlexanderP, 2019-12-02:
> What about Debian stretch?
Build was successful:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9bfd04d0..9bede12c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -7,8 +7,10 @@ if(CMAKE_SOURCE_DIR STREQUAL CMAKE_BINARY_DIR)
             "    mkdir build && cd build && cmake ..")
 endif()
 
-cmake_minimum_required(VERSION 3.2.0 FATAL_ERROR)
-cmake_policy(VERSION 3.11)
+cmake_minimum_required(VERSION 3.7.2 FATAL_ERROR)
+if(NOT CMAKE_VERSION VERSION_LESS 3.11.0)
+    cmake_policy(VERSION 3.11.0)
+endif()
 set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}
     "${CMAKE_SOURCE_DIR}/cmake/")
 set(CMAKE_CXX_STANDARD 11)
```

whitequark, 2019-12-02:
Fixed in 0b786e6e2.

closed 2019-12-02 by whitequark