solvespace/solvespace #521: Z order for selecting coplanar entities is inverted

Symbian9, 2019-12-01:
### System information

- SolveSpace version: `3.0~690f87c`
- Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

If try to select one of co-planar entities, SolveSpace should prioritize to select nearest entity (according display Z axis).

### Actual behavior

When tried to select one of co-planar entities, SolveSpace selected remote in space entity (according display Z axis), instead of selecting nearest entity.

[image]

[image]

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2019-12-01:
Related to #161. I recall there was a lot of trouble getting selection logic to work correctly.

Evil-Spirit, 2020-05-13:
Need to integrate "Selection by order" where I implemented to following. 
1. When clicking, create a list of constraints/entities inside click radius.
2. See if some of them is selected.
3. Select the next entitiy after selected when click

This allows to select any entity even if them co-planar.

ruevs, 2020-05-13:
> Need to integrate "Selection by order" where I implemented to following:

I can not find it. Or do you mean in NoteCAD?

Evil-Spirit, 2020-05-13:
This one
https://github.com/Evil-Spirit/solvespace/commit/a7be01991d4d30556d1442d5d3c5da13b67bcbe9

whitequark, 2020-05-13:
IIRC "selection by order" solved this problem but introduced new ones.

phkahler, 2020-05-23:
I also get frustrated at times when trying to drag a point and it grabs planes from g001-references and won't move. Perhaps those should only be selected if it's unambiguous?

whitequark, 2020-05-23:
We'll get the exact same problem in other groups if we special-case references. I think this can and should be solved properly instead.

Evil-Spirit, 2020-05-23:
Problem with coplanar entities can't be solved, so selection by order is common workaround in other software.

whitequark, 2020-05-23:
Right, we can do that, and we can also make it possible to disambiguate the selection somehow (like what KiCAD does).

phkahler, 2020-10-29:
I think it should prefer entities in the current group, Then entities (points) that lie in the current workplane, and then others.

phkahler, 2020-10-31:
Reopened. The recent fix is not very robust. Better but still not great.

phkahler, 2020-11-01:
I really wanted to get this fixed for 3.0 and gave it a try. However, if I read the code correctly and this definition:

```
    // A hovered entity, with its location relative to the cursor.
    class Hover {
    public:
        int         zIndex;
        double      distance;
        Selection   selection;
    };
```
I believe distance is the lateral distance from the mouse pointer to an entity, while zIndex is a terrible name because it's based on the style of the entity which differentiates points, curves, and constraints. There is nothing that corresponds to the depth into the screen so proper Z-ordering is not just broken, it doesn't exist. This is great for 2D sketches but obviously has some issues in 3D.

I'm thinking a correct fix is going to add a "depth" member to the Hover, a bit of new calculations in all the picking code to provide the depth value for each entity, and then finally changes to ChooseFromHoverToSelect() and ChooseFromHoverToDrag(). And lastly verify that grid snapping still works!

closed 2020-11-17 by phkahler

ruevs, 2020-11-18:
Works fine now. And I have not noticed any regressions.

PR #787 (2020-10-31): Use zOrder for selections within the same group
Fixes #521