solvespace/solvespace #522: "Glass effect" is broken

Symbian9, 2019-12-03:
### System information

- SolveSpace version: 3.0~690f87c
- Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

Opaque solids should be shown throw half-transparent solid as throw "glass".

### Actual behavior

Mesh faces of opaque solids inverted under half-transparent solid

[image]

In same time another half-transparent solid, placed under half-transparent, is fully invisible

[image]

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2019-12-03:
This has been discussed several years ago and there is no simple way to support high-quality transparency (you have to carefully sort solids before rendering them). Fixing this will require redesigning the entire rendering pipeline, which is an amount of effort inordinate with the usefulness of the transparency feature. Therefore this will not be fixed.

Evil-Spirit, 2019-12-07:
We can make reordering of opacity materials to be in the last place in mesh. Also, we can switch off zwrite for opacity objects and probaby switch off backface culling for them.

whitequark, 2019-12-07:
If I had known when introducing it that transparent meshes were so complicated I would not have done it in the first place. (I understood very little about graphics programming at the time.) I certainly have never intended for this feature to eventually require a major rearrangement of the rendering code...

Symbian9, 2019-12-08:
As I know, @leozide did a lot of work in *LeoCAD* for correct render mesh in same situation.

- https://github.com/leozide/leocad/commit/081373742ee503cfd10bcfb892d536156e52ac51

[image]

ruevs, 2021-03-29:
Reopening because of #977