solvespace/solvespace #523: Export canvas x / y offset is inverted

Symbian9, 2019-12-03:
### System information

- SolveSpace version: `3.0~690f87c`
- Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

Export canvas x / y offset should be relative to left-bottom x / y position.

### Actual behavior

Export canvas x / y offset actually is inverted (really it should be -x / -y, as actually positive increasing offsets move export canvas origin to down-left direction)

[image]

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

phkahler, 2020-06-25:
@Symbian9 is this right? It seems like the offset should be relative to the canvas origin. If you export with no offset it would place things at the edge of the canvas, so you'd want to use a positive offset within that coordinate system. At least that's my interpretation which could be wrong. Same thing when 3d printing, I have to tell the slicer where my print-bed corner is relative to the home position of the printer - which would look a lot like this and requires positive offset values.

Symbian9, 2020-06-25:
> is this right? It seems like the offset should be relative to the canvas origin.

Yes.

@phkahler, on my screenshot above *lower left corner* of "rectangle" entity placed at the canvas origin (by *Center View At Point"*).

But if set *"export canvas size: `fixed`"* and set `x`/`y` offset properties to *positive values*, export canvas moved to ⭹ direction (instead of expected moving to ⭷ direction).

So, actually "entities" moved according offset instead of move "export canvas".

SOLUTION: Switch internal calculated positive values of `x`/`y` offset properties to negative values *only* for *"export canvas size: `fixed`"* option.

phkahler, 2020-06-25:
So here is a picture. I created a 50x40 rectangle with the lower point at the origin. I exported as svg and opened that at the top. It's hard to see the purple because there is no background. The X and Y offsets work exactly as I expect. The rectangle is 20 from the left of the SVG origin and 5 from the bottom - as specified in the export offset values. I also confirmed that 0 aligns with the left side for X offset and bottom for Y. I honestly don't see any problem here.

I also tried exporting G-code, but it didn't seem to be affected by these settings - the point coordinates were 0, 40 , and 50.  I have not tried any other export options. As far as I can tell these settings are not visible in any way on the SolveSpace sketch and I would not expect them to be. @Symbian9 Can you be give a better example of the problem? Perhaps a simple .slvs and an exported file that behaves incorrectly?

[image]

Symbian9, 2020-06-25:
> The rectangle is 20 from the left of the SVG origin and 5 from the bottom - as specified in the export offset values.

But for *"export canvas size: `fixed`"* optioon the offset should be *relative to the canvas origin*; this is NOT "paper margins" (as it defined for "export canvas size: `auto`" option).

> I have not tried any other export options.

Try export 2D view to PDF.

P.S. I will add more screenshots (or video) of my case later today.

phkahler, 2020-06-25:
I tried one as "auto" and it seems to work as indicated "(by margins around exported geometry)" where the exported SVG contains only the actual geometry plus margins as specified.

[image]

That example was not at the origin and I think I get what you're after.

When you say "for "export canvas size: fixed" option the "offset x/y" should be relative to the canvas origin." do you mean the solvespace sketch origin? Do you want to specifiy in SS coordinates the lower left corner of the exported area and its width and height?

I'm trying that and getting really confused. To get this:
[image]

I had to specify negative offsets, which works kind of like I expected. The geometry was offset by the amount specified. What you want is for the -6 and -24 to be positive 6 and 24 is that correct? If so, I would suggest that the wording not be "offset" but "coordinates" of the exported area. Weather that warrants another option I don't know.

What I found really disturbing is that even with the negative numbers it didn't work either until I centered the view on the sketch origin. I don't think the view should have any influence on the exported image. IMHO this behavior is not discoverable or up to interpretation of wording about coordinates and offsets. The only reason I even tried it is because @Symbian9 indicated doing so in the original problem statement here.

I hope I've clarified more than confused anyone reading.

Symbian9, 2020-06-25:
> When you say "for "export canvas size: fixed" option the "offset x/y" should be relative to the canvas origin." do you mean the solvespace sketch origin?

No, I mean working space canvas origin (as defined in "Center View At Point" option).

phkahler, 2020-09-16:
@Symbian9  Can you verify this change does what you want for this issue?
https://github.com/phkahler/solvespace/tree/issue523

It's just the last commit: https://github.com/phkahler/solvespace/commit/d94cfade9d421d1d7945ee5b289b2fcb28e97ec1

closed 2020-09-20 by phkahler