solvespace/solvespace #524: Flat colors render mode

Symbian9, 2019-12-03:
### System information

- SolveSpace version: `3.0~690f87c`
- Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

Add an option to enable "flat colors" render:  (direct RGB values without any shadings/gradients; independent of light direction)

### Actual behavior

Actual render has shading/gradients, according to default "light direction" setting.

[image]

[image]

And it's non-trivial task to get near "flat colors" render in SolveSpace.

[image]

As workaround I use (`10000` used because with `10`...`1000` shading/gradients steel appear):

```
light direction                 intensity     
  #0    (0.00, 0.00, 10000.00)[c]  1.00 [c]
  #1    ( 0.00, 0.00, 0.00)[c]  0.50 [c]
```

[image]

### Additional information

As side note, seems like there no way to reset "light direction" setting to default values:

```
light direction                 intensity     
  #0    (-1.00, 1.00, 0.00)[c]  1.00 [c]
  #1    ( 1.00, 0.00, 0.00)[c]  0.50 [c]
```

So, it would be good to add button for reset "light direction" too.

[image]

Symbian9, 2019-12-03:
Think, such mode would be especially useful for export 2D view to PDF and SVG.

Symbian9, 2020-05-14:
I just wondered when tried to place reference image behind solid model while *"Hide shaded view of solid model"* is toggled — it looks like "flat render" which just use background color for triangle mesh of this solid model.

If so, maybe this is the start point to *"Show flat color view of solid model"*?

[image]

phkahler, 2020-07-13:
One option is to set the intensity of both light sources to 0.0.   There is still an ambient light which will show the model but will not be as bright as you may want (white shows as a uniform grey). Proper flat shading might just be turning off the existing lights and changing the ambient brightness which is no a UI option.

phkahler, 2020-07-13:
@Symbian9  Digging all the way down to the shader code it looks like simply setting the ambient lighting to 1.0 would effectively disable the 2 lights and produce the desired effect. Unfortunately that parameter isn't in the UI at the moment.

Symbian9, 2020-07-13:
>  simply setting the ambient lighting to 1.0 would effectively disable the 2 lights and produce the desired effect. Unfortunately that parameter isn't in the UI at the moment.

Nice.

@phkahler, I hope, you would be able to solve this issue & add some UI controls for enhanced control for render parameters into "Property Browser".

phkahler, 2020-07-15:
@whitequark what would you think of automatically setting the ambient lighting to 1.0 when both lights are set to zero brightness? That would not involve a UI change but should provide flat coloring if wanted.

@Evil-Spirit Related to shading, what is the coordinate system for the lights? I thought the position was x,y,z with +x to the right, +y up and +z out of the screen. But when mucking about with the shader code it seems there might be some sign flipping going on between the UI and the shader, is that correct?

whitequark, 2020-07-15:
> what would you think of automatically setting the ambient lighting to 1.0 when both lights are set to zero brightness?

That seems really hard to discover. I think a checkbox like `[✔] Flat shading` that hides the configuration for the lights if set would work fine.

Evil-Spirit, 2020-07-16:
@phkahler 
> Related to shading, what is the coordinate system for the lights

Actually don't remember. One of the lights direction probably is in camera space.

Symbian9, 2020-07-16:
> what is wrong with your solution to place the light at 10,000? Is it just the lack of any way to reset it to normal without writing it down or remembering?

Beside it looks as needed on screen & on export 2D view to PNG, placing the light at Z: `10000` would not provide "truly" *flat colors* & on export 2D view into PDF & SVG with  "🗹 `export shaded 2D triangles` it still exported with *shaded colors*.

[image]

So my solution with `10000` could not fully solve this issue.

> That seems really hard to discover. I think a checkbox like `🗹 Flat shading` that hides the configuration for the lights if set would work fine. The implementation could be the same otherwise.

@whitequark, it would be great! Also, on implementing this solution add related CLI option too, please!

phkahler, 2020-07-17:
> I think a checkbox like `[✔] Flat shading` that hides the configuration for the lights if set would work fine.

@whitequark I'm leaning more towards adding the Ambient parameter to the lighting config. I've also been experimenting with shaders and may have another parameter at some point.

closed 2020-07-18 by phkahler