solvespace/solvespace #526: SolveSpace 3.0 icon improvements

Symbian9, 2019-12-04:
### System information

-  SolveSpace version: `3.0~690f87c`
-  Operating system: Debian 9.x stretch (x86_64)

### Expected behavior

Propose to improve *SolveSpace* `3.0` icon

- add grey dot for "connected points"
- replace transparent background with opaque black (as SolveSpace default render setting)
- invert "line segments" color from black to white (as SolveSpace default render setting)
- remove shade under "constraint perpendicular" icon

[image]

### Actual behavior

Actual *SolveSpace* `3.0` icon include shade under "constraint perpendicular" icon

- https://github.com/solvespace/solvespace/blob/master/res/freedesktop/solvespace-48x48.png

[image]

Also, its SVG version has incorrect order for "green dot" and "line segments" shapes

[image]

### Additional information

- Proposed `solvespace-scalable.svg` - https://gist.github.com/Symbian9/5f8dad142045cd374d585acfac0d4cdd

whitequark, 2019-12-04:
An icon is a symbol. It does not have to look exactly like a screenshot.

I've fixed the Z-order issue in SVG in 69baf524.

closed 2019-12-04 by whitequark

Symbian9, 2019-12-04:
BTW, Position of shade under "constraint perpendicular" icon on PNG version differ from SVG.