solvespace/solvespace #527: linux mingw compile error: `flatc.exe: Exec format error`

avh-on1, 2019-12-09:
### System information

SolveSpace version: 3.0~64c0f62b

Operating system: Debian Stable (Buster) amd64

### Expected behavior

Following the readme's windows cross-compile instructions results in solvespace.exe being built.

### Actual behavior
The build fails, with the following error:

```
Scanning dependencies of target q3d_header
[ 13%] Generating q3d_object_generated.h
/bin/sh: 1: ../../bin/flatc.exe: Exec format error
make[2]: *** [extlib/q3d/CMakeFiles/q3d_header.dir/build.make:61: extlib/q3d/q3d_object_generated.h] Error 2
make[1]: *** [CMakeFiles/Makefile2:230: extlib/q3d/CMakeFiles/q3d_header.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

```

### Additional information

This happens when following the readme's instructions for 32-bit and 64-bit windows builds.

Linux builds on the same machine work fine.

whitequark, 2019-12-09:
Hm, right, this broke after #384. You'll need either wine-binfmt or flatc installed on the host system and specified with `-DFLATC=<path-to-flatc>` when you invoke CMake.

avh-on1, 2019-12-11:
I installed `wine-binfmt` and re-ran the build. It worked without any other problems. Thanks!

closed 2019-12-11 by whitequark

outdooracorn, 2020-04-27:
I've just run into this issue too. Installing `wine-binfmt` didn't solve the problem (`flatc` was not installed) for me in an `ubuntu:20.04` docker container.

For anyone else interested, I had to install `libflatbuffers-dev`, rerun `cmake` as before but with the extra option `-DFLATC=/usr/bin/flatc` and rerun `make`. Thanks @whitequark for the instructions!

Should these additional steps be added to the README?

whitequark, 2020-04-27:
Done: 35448298

miljbee, 2020-05-19:
Hello,

I am having another issue, but I think it's also related to libflatbuffers-dev.
Problem is apt-get install libflatbuffers-dev = E: Unable to locate package libflatbuffers-dev

I am also having an error with OpenMP. I have no idea what it means !

OS is Ubuntu 18 on Win !
Ubuntu 18.04.1 LTS

I am trying to build an up2date windows version. Thanks for your help !

~/solvespace/build$ cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchain-mingw64.cmake -DCMAKE_BUILD_TYPE=Release -DFLATC=$(which flatc)
-- Found OpenMP_C: -fopenmp
-- Found OpenMP_CXX: -fopenmp
-- Found OpenMP: TRUE
CMake Warning at CMakeLists.txt:105 (message):
  OpenMP not found, geometric operations will be single-threaded

-- Using in-tree libdxfrw
-- Using in-tree flatbuffers
-- Using toolchain file: ../cmake/Toolchain-mingw64.cmake.
-- Using in-tree q3d
CMake Error at CMakeLists.txt:175 (add_subdirectory):
  The source directory

    /home/miljbee/solvespace/extlib/q3d

  does not contain a CMakeLists.txt file.

-- Using in-tree zlib
-- Found ZLIB: zlibstatic (found version "1.2.9")
-- Using in-tree libpng
-- Found ZLIB: zlibstatic
-- Using in-tree freetype
-- Creating directory /home/miljbee/solvespace/build/extlib/freetype/include/freetype/config
-- Creating file /home/miljbee/solvespace/build/extlib/freetype/include/freetype/config/ftconfig.h
-- Creating file /home/miljbee/solvespace/build/extlib/freetype/include/freetype/config/ftoption.h
-- Using in-tree pixman
-- Building pixman version 0.34.1
-- Found OpenMP_C: -fopenmp
-- Found OpenMP_CXX: -fopenmp
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Using in-tree cairo
-- Building cairo version 1.15.2
-- Using in-tree ANGLE
CMake Warning at CMakeLists.txt:303 (message):
  Gettext not found, translations will not be updated

whitequark, 2020-05-19:
> I am trying to build an up2date windows version.

You're building on Windows, right? Why not use MSVC?

miljbee, 2020-05-19:
> > I am trying to build an up2date windows version.
> 
> You're building on Windows, right? Why not use MSVC?

Because I wanted to try that way !

whitequark, 2020-05-19:
Ok.

> Problem is apt-get install libflatbuffers-dev = E: Unable to locate package libflatbuffers-dev

Ubuntu 18.04 doesn't have flatbuffers in the repository. Can you use Ubuntu 20.04? That's the up-to-date LTS release, so I assume there should be no issues with it.

> I am also having an error with OpenMP. I have no idea what it means !

Some recent performance improvements require OpenMP. However, MinGW compilers shipped with Debian/Ubuntu (really, most MinGW compilers shipped by anyone) don't support it. So your performance will be the same as it was before.

miljbee, 2020-05-20:
Thanks for your help. Unfortunately Ubuntu can't be upgraded to 20 on win for now. 
So I downloaded VS, and it was so easy to build ... Just entered the git address, VS downloaded the code and ran cmake automatically, I just had to hit build ... and got my solvespace.exe. The exe file is big (15MO) I think it might be a debug version. I have to find how to switch to release.

By the way : awesome software, I love it, thanks for your work.

whitequark, 2020-05-21:
Sweet! I'm glad it was easy for you and that you like it.