solvespace/solvespace #534: current version has performance regressions

prototype99, 2019-12-22:
### System information

SolveSpace version: 3.0~dcdfdec5

Operating system: Gentoo 17.0

### Expected behavior

Performance is the same as, or better than version 2.3

### Actual behavior

most changes seem to take a lot longer, although changing the camera is still snappy.

### Additional information

[room-plan.zip](https://github.com/solvespace/solvespace/files/3993117/room-plan.zip)

whitequark, 2019-12-23:
Are you building SolveSpace yourself? Are you building a release version?

Symbian9, 2019-12-23:
> [image]

According its screenshot FPS reached `1675` 1/s. What is performance with *SolveSpace* `2.3` on same scene on your PC?

Could you also give more details on your configuration

[image]

Also, seems like few group has not hard constrained (1 or more DOF in groups)

[image]

P.S.: As for me, latest nightly builds of *SolveSpace* `3.x` (maybe, `3.0~690f87c`) for Debian 9.x has much better performance than previous `3.x` builds and `2.3` release

prototype99, 2019-12-23:
> Are you building SolveSpace yourself? Are you building a release version?

building myself, I used CMAKE_BUILD_TYPE="Release" in case that makes a big difference. the commit used is based on the last time the changelog was updated, with the intention to go back later and see if there's any later commits I may want to give a try if I have any issues.

> > (screenshot)
> 
> According its screenshot FPS reached `1675` 1/s. What is performance with _SolveSpace_ `2.3` on same scene on your PC?

That seems to be more the fps for displaying the model, that has always been smooth, but in 2.3 dragging models around is smooth whereas in the 3.x version I used, not so much. same goes for sketching in 3d and dragging unconstrained lines in sketches.

> Could you also give more details on your configuration

Model:			Dell Latitude E6330 laptop system.
Ram:			8GB DDR3
CPU:			Intel "ivybridge" i5-3320M
Opengl vendor:	Intel
Renderer:		Mesa DRI Intel Ivybridge mobile
Version:		4.2 (core profile) mesa 19.1.8

> Also, seems like few group has not hard constrained (1 or more DOF in groups)

the current assembly objects I'm working on, so it's expected

> P.S.: As for me, latest nightly builds of _SolveSpace_ `3.x` (maybe, `3.0~690f87c`) for Debian 9.x has much better performance than previous `3.x` builds and `2.3` release

That one is older than my version, but I will check it anyway, if it performs better that would at least narrow down when performance changed

Symbian9, 2019-12-23:
> > Also, seems like few group has not hard constrained (1 or more DOF in groups)
>
> the current assembly objects I'm working on, so it's expected

To be honest, in SolveSpace (from my personal experience) you should hard constrain all entities in all previous groups and work with only last group.

Symbian9, 2019-12-23:
>  > P.S.: As for me, latest nightly builds of SolveSpace 3.x (maybe, 3.0~690f87c)

> That one is older than my version, but I will check it anyway,

I should be clear here: all new nightly builds **since** `3.0~690f87c`

> latest nightly builds of SolveSpace 3.x (maybe, **since** 3.0~690f87c) for Debian 9.x has much better performance than previous 3.x builds and 2.3 release

prototype99, 2019-12-23:
> I should be clear here: all new nightly builds **since** `3.0~690f87c`

so potentially it's specific to certain hardware, or due to a difference in how it's built?

phkahler, 2020-10-01:
@prototype99  Did you ever find out where the performance regression came from?

Also since you're building from source there have been some improvements. Try the following with master:
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_OPENMP=yes -DENABLE_LTO=yes
```
LTO will take a lot longer to build but gives another boost.

closed 2020-10-19 by phkahler