solvespace/solvespace #536: "Edit newly added dimensions" tries to allow editing auto-reference dimensions

rpavlik, 2019-12-23:
### System information

SolveSpace version: 3.0~64c0f62b

Operating system: Debian Buster

### Expected behavior

I turned on "Edit newly added dimensions" because it's a great workflow improvement.

When adding a distance constraint on an already fully-constrained sketch, I'd expect it to not let me edit it, since solvespace will automatically mark such a constraint as "REF"

### Actual behavior

It lets you "edit" it, though the edits have no effect.

### Additional information

[image]

[image]

Here's the model for the second one, 
[example.zip](https://github.com/solvespace/solvespace/files/3995464/example.zip)

vespakoen, 2020-01-22:
I added that functionality and didn't think of this case.
I will try to make a PR when I've got some time on my hands

whitequark, 2020-01-23:
Thanks for following up!

rpavlik, 2020-03-07:
Looks fixed since https://github.com/solvespace/solvespace/commit/3ed1719de2dc6e0525c94de78353d076623e46e4, thanks!

closed 2020-03-07 by rpavlik