solvespace/solvespace #538: RFE: Default "export" filename stem and directory to same as slvs file

rpavlik, 2019-12-23:
Just a UI nitpick. If I get the chance I'll look into doing this myself.

### System information

SolveSpace version: 3.0~64c0f62b

Operating system: Debian Buster

### Expected behavior

I have a solvespace file. I go to export it to trimesh (to get an STL to 3d print). The dialog probably should open to the location of the slvs file and defaulting to the same stem but with the appropriate extension (`.stl`)

### Actual behavior

It defaults to "untitled" (`untitled.stl`) in some other directory - think it's the last directory I used that export option in.

whitequark, 2019-12-24:
> It defaults to "untitled" (`untitled.stl`) in some other directory - think it's the last directory I used that export option in.

That's on purpose: suppose you are exporting many files in some other directory (like "Outputs" in the root of your projects).

rpavlik, 2019-12-31:
ah, so it's a workflow problem on my part that I tend to have my stl files live next to the slvs files and edit one model at a time :) 

Would defaulting the name to match the model make sense, though?

whitequark, 2019-12-31:
Yep. That's how it worked before 6b5db5897155ca9bdc5daa3e791c07723a6b7d9c; it looks like I forgot to add functionality that sets the file stem. You should be able to use `Platform::Path` and `FileDialog::SetCurrentName` to add this back easily.

whitequark, 2019-12-31:
Regarding your workflow, I would expect you to use solvespace-cli, seeing as you're (I assume) comfortable with command line.

commit e59186a4 (2020-12-10): Suggest sensible defaults in file dialogs

Went through first the diff of the referenced commit,
then all instances of "Create(Open|Save)FileDialog";
added SuggestFilename() calls where a default exists

This has been previously removed in
6b5db5897155ca9bdc5daa3e791c07723a6b7d9c
Closes #538
closed 2020-12-10 by phkahler