solvespace/solvespace #540: Weird rendering of solid difference

baryluk, 2020-01-13:
### System information

Compiled at commit 16c5fa6889202f2059b52c4a89599910541dd1a0

Operating system: Debian testing, amd64

### Expected behavior

What should have happened?

I had a lathe operation, plus 2 extra extrusions (one with 3x time rotating symmetry) acting as a difference.

At the end I wanted to see a crossection, so I did another sketch, extruded it, and used as a difference.

I expected to have a solid model, not hollow one, with some red marks on the edges. There are some other red marks, that I don't know what they are, and some edges (45 deg ones) do not have the red edges.

No idea what is going on.

### Actual behavior

[image]

[image]

Project file:  [adapter-plate.tar.gz](https://github.com/solvespace/solvespace/files/4054379/adapter-plate.tar.gz)

baryluk, 2020-01-13:
I just tried using https://github.com/phkahler/solvespace/tree/nurbs branch, and result is the same or even worse:

[image]

whitequark, 2020-01-13:
Classic NURBS issue. Force triangle mesh for now.

baryluk, 2020-01-13:
@whitequark  Indeed that helped:

[image]

How will that affect STEP 3D export?

whitequark, 2020-01-13:
You wouldn't be able to export rational surfaces after this, only triangle meshes; in particular IIRC exported STEP files will just be empty and you'll have to use STL.

baryluk, 2020-01-13:
@whitequark That sucks. :/ What if I keep the model with this red things / errors and export then to STEP? Is the red thing due to rendering issue, or the model itself is broken due to intersection problems?

whitequark, 2020-01-13:
The model itself is broken. If you *really* need NURBS surfaces, try making the intersection not "cut across the seam". E.g. if you have a difference cutting exactly 1/4 of a circle, try moving it away by 0.1 mm or something.

baryluk, 2020-01-13:
@whitequark Thanks a lot.

PR #733 (2020-10-10): Create intersection curves from existing ones.
When a plane coinsides with a seam we need to copy that trim curve. The existing curve belongs to the original shell surfaces and an intersection is otherwise not found. Closes #540.
commit d72eba80 (2020-10-10): Create intersection curves from existing ones.

When a plane coinsides with a seam we need to copy that trim curve. The existing curve belongs to the original shell surfaces and an intersection is otherwise not found. Fixes #540.
closed 2020-10-10 by phkahler