solvespace/solvespace #542: Algebraic constraints

baryluk, 2020-01-16:
This is a feature request.

I would like to be able to add linear and non-linear constraints. Either as a side note on the sketch, with a symbolic names referencing dimensions, or for some constraints of the form: 'B = something_without_B' directly in the constrain line segment/distance/angle/radius.

[image]

Ability to add implicit constraints between dimensions (ones which can't be inverted algebraically) would also be really cool:

[image]

whitequark, 2020-01-16:
Duplicate of #77, #81, etc.

closed 2020-01-16 by whitequark

baryluk, 2020-01-16:
@whitequark Sorry for the duplicate and thanks for pointing me to the existing bugs. I did a search before posting, but I didn't see these bugs.

Cheers.