solvespace/solvespace #550: Helix tool almost always results in invalid mesh

andres-erbsen, 2020-02-09:
### System information

dcc80de on Arch Linux

### Expected behavior

A manifold mesh representing the helical extrusion, for example

[image]

### Actual behavior

If I increase the angle just a bit I get a self-intersecting mesh:

[image]

and increasing the angle just slightly more gives naked edges instead:

[image]

### Additional information

Note that the issues are present even with "force NURBS surfaces to triangle mesh".

This example (unit square at unit distance from axis): [helix.zip](https://github.com/solvespace/solvespace/files/4176517/helix.zip)
Non-trivial example (8x rotated 2 beziers approximating involute that is almost a triangle): [gear90.zip](https://github.com/solvespace/solvespace/files/4176518/gear90.zip)
This also occurse when applying helix to a circle.

#472 seems relevant.

phkahler, 2020-02-09:
The problems with helix.zip seem to be solved by #473 which is very helpful for extrusions with a lot of twist. No naked edges.

Basic rendering with gear.zip is also better with those changes. There are fewer errors when using Analyze -> show naked edges but there are still a lot of them with that model. I'm not sure what causes them in that case - lowering the chord tolerance causes more of them.

mkyral, 2020-05-21:
Hi,
I've played with Helix (really useful tool) in and I've generated this problematic shape: [rounded_edges_issue.zip](https://github.com/solvespace/solvespace/files/4664827/rounded_edges_issue.zip)

[image]
[image]
[image]

phkahler, 2020-05-22:
@mkyral  You can fix the naked edges problem (sort of, I hate the need to do this) by clicking on g005-helix in the text window. Then check the box for "force NURBS surfaces to triangle mesh. SolveSpace has some NURBS issues, but NURBS are preferable to meshes so long as those issues don't show up. I tried a reconstructing the curve sketch differently but ran into the same issue.

A suggestion about the design. If you're not going to slide the opening to the side, use a reVolve instead of a Helix. It doesn't solve the issue here, but SolveSpace then recognizes the flat surfaces on the sides as flat (you can select them and constrain things to them). Also go to home->configuration->chord tolerance and lower that value to 0.1 it will look a little better on-screen (lowering that too much will degrade performance).

mkyral, 2020-05-22:
Hi,
Thanks for your suggestions. ReVolve works better in this case.

phkahler, 2020-07-03:
While experimenting with variations on grid triangulation I noticed that changes to the interior grid can affect the number of naked edges on that gear90 model. This has me thinking it's an issue with ear triangulation on a helix, since the grid has no effect on outer edges.

Speculations: maybe a problem with very narrow triangles, or near nonlinear segments.

phkahler, 2020-07-08:
@andres-erbsen The rest of this is fixed in PR#643 The gear90 model will have no naked edges if chord tolerance is set to either 0.5 or 0.2.  They do reappear when its set to 0.1 though so it's not the ultimate fix, just a good improvement.

I am curious why you made the gear teeth with that type of curve. Would an arc be OK? How about one cubic bezier for each side of the tooth? Those would create more uniform curve segments which may help. I'm not a gear expert so I don't know what's more accurate.

andres-erbsen, 2020-07-08:
Thanks!

I'm no expert either; I think I was following [this tutorial](https://www.youtube.com/watch?v=i6tDWJsNsok) for the bezier approximation of the common [involute gear](https://en.wikipedia.org/wiki/Involute_gear) profile. I tried the same with a single cubic bezier just now and I was able to get it within fabrication tolerance of what I want in this example. The linked construction is neat, though, because it doesn't require eyeballing the locations of the control points (but doing this construction with just two control points gives a straight line).

closed 2020-07-11 by phkahler