solvespace/solvespace #552: Roadmap?

DougBeney, 2020-02-13:
Hi!

Curious if there is a release roadmap for this great piece of software.

There seems to be plenty of frequent commits but there hasn't been a new release for the past 4 years.

Thanks!

chipolux, 2020-02-24:
I've been wondering the same thing, I've been building from source on macOS recently but it would be nice to have a ballpark of when the next release would be or at least what features/bugs are blocking one.

whitequark, 2020-02-24:
I'm broke and chronically ill. The next release will happen when it happens. (Or, there is a number of other people who have write access to the repository who could also cut a release--I'm not exactly positioning myself as a blocker here--but I think they're pretty busy too.)

Evil-Spirit, 2020-02-25:
> I'm broke and chronically ill.

Please please get well soon.

phkahler, 2020-03-04:
The last time this came up, whitequark indicated a desire to make a release after Helix/Revolve got in. While those features have landed, I wouldn't consider them adequate until #473 gets merged which is on my plate. I'm sorry, I've been neglecting SolveSpace for personal reasons since September. I hope to get that PR cleaned up and merged soon.

There are some other related things that would be nice like icons for those new tools - anyone want to draw them? I'm serious. Someone make appropriate icons (they're layered bitmaps) and I'll tweak the code to include them on the toolbar.

I was also really hoping #35 could get done for the next release as well. It was working even though it still used the "dual difference" method for triangle meshes. I can only assume @ruevs is also busy with other things. That work was impressive and it'd be a nice feature and close one of the oldest open issue.

Obviously we'd like all the open issues closed and all our favorite features added prior to a release, but that isn't going to happen. I just mention those above because I feel they're close enough to make a quick sprint to complete and will give users some nice things on top of all the other great things that have gone in since 2.3.

whitequark, 2020-03-05:
> Someone make appropriate icons (they're layered bitmaps) and I'll tweak the code to include them on the toolbar.

Ideally the new icons should be SVGs so that we can generate @2x versions of them to use on HiDPI screens. But it's kind of hard to make SVGs that look nice on both @1x and @2x. It might be easier to just draw them twice, but I can't ask unpaid contributors to do double the work just so that icons look a bit better.

rpavlik, 2020-03-07:
I recently tried using 2.3 to check the differences, and I forgot how much better 3.0 is at automatically removing redundant constraints. The workflow is so much smoother. I'd almost be in favor of temporarily disabling features that might not be quite ready (if any) just to get those changes out there. If there's something I can do to help with release, I'd be happy to. (I can do some icons, I bet....) Got a few "upstream contribution" hours a week at work to chip in too.

whitequark, 2020-03-08:
> I'd almost be in favor of temporarily disabling features that might not be quite ready (if any) just to get those changes out there.

I agree. Let's give @phkahler some time to update the PRs (I'd normally just update it myself before merging but ... not well lately) and then just cut 3.0. I think current SolveSpace master is both very reliable in absolute terms, and also works a lot better than the bar most people have for an mCAD. So, while there are open bugs and many features we'd like to see, let's not wait for Godot and just do a release at last.

ruevs, 2020-03-13:
You are right - I have not gone back to intersection in a while... :-(

Intersection in NURBS mode works fine here:
https://github.com/ruevs/solvespace/tree/wip-intersection

However the code is very dirty and full of  debug stuff because:

1. I still have not figured out how to flip (make the edges "go around" in the opposite direction) some of the surfaces needed for the intersection. So that they end up with normals pointing out. Thus this ugly hack is in:
https://github.com/ruevs/solvespace/commit/7b09fddcf5e66d4fc1c8ee5efe93f530e042555c
see @jwesthues Jonathans' comment here:
https://github.com/solvespace/solvespace/issues/35#issuecomment-531173543

2. I have not spent any time figuring out how to do intersection in triangle mesh mode.
This this commit:
https://github.com/ruevs/solvespace/commit/d16c4e8554d1a87b43353bee30e4366c7c8abf6d#diff-9a5625e73a45d7ece1335647e0fa27d6L289
that uses the "C=A-(A-B)" method that @ghoss originally used..

3. Found some broken corner cases in difference that I think I can fix if I spend enough time.

phkahler, 2020-03-29:
My helix nurbs fixes are in for 3.0 so it's releasable.

whitequark, 2020-03-29:
Great. Give me a few weeks to go through the accumulated issues / PRs (I have even less time than usual, being stuck in an international move during a pandemic) and I'll cut a release. Or if I for some reason never get around, you have all the credentials you need to do it yourself.

phkahler, 2020-03-30:
@whitequark can you outline what needs to happen? Also, is there a release procedure? I'm happy to help, but this is new to me.

@chipolux what version of MacOS are you running on? I thought that was being dropped due to OpenGL issues?

whitequark, 2020-03-31:
> can you outline what needs to happen?

What needs to happen is going through the stack of pull requests and issues marked with `patch-available` and figuring out which, if any, should go to a release. You can't help much precisely because you're new to this codebase; often, seemingly trivial chances violate invariants that aren't documented anywhere, which is why reviews are so laborous.

ruevs, 2020-05-08:
Intersection is done:
https://github.com/solvespace/solvespace/pull/595
https://github.com/solvespace/solvespace/issues/35

chipolux, 2020-05-13:
> @chipolux what version of MacOS are you running on? I thought that was being dropped due to OpenGL issues?

@phkahler, currently running on Catalina 10.15.4 and everything seems to work correctly. I'm also fine with continuing to build from source for myself.

[image]

@whitequark @ruevs thank all you guys for creating such an awesome piece of software! It's awesome at letting me get my little hobby projects done without getting in my way or dragging my machine to a crawl. So no pressure from me! I'm just excited to recommend it to more of my DIY buddies that aren't quite ready to build from source.

ruevs, 2020-05-13:
>
> 
> @whitequark @ruevs thank all you guys for creating such an awesome piece of software!

You do not need to thank me - my contribution is tiny.

>
>
> So no pressure from me! I'm just excited to recommend it to more of my DIY buddies that aren't quite ready to build from source.

That's understandable. You could do like me:
There is continuous integration (Solvespace is built for Windows, macOS and Linux) for every pull request. So you go here:
https://github.com/solvespace/solvespace/pulls?q=is%3Apr+is%3Aclosed
Finds a recent _closed_ pull request, browse to the "artifacts" for the OS they use and send them the link. The artifacts can be a bit tricky to find - you nerd to find the "View Details" button and click it:
[image]

whitequark, 2020-05-13:
@ruevs You must be thinking about artifacts published on Appveyor. There are no such artifacts published on Travis, so if you use macOS you can only build from source.

ruevs, 2020-05-13:
Oh sorry, I have indeed given links only to Windows versions, no one has ever asked me for a Mac version. I just presumed Travis did the same.

whitequark, 2020-05-13:
Nope, there's no way to do this with Travis, sadly.

Symbian9, 2020-05-14:
> > > You must be thinking about artifacts published on Appveyor. There are no such artifacts published on Travis, so if you use macOS you can only build from source.
> >
> > I just presumed Travis did the same.
>
> Nope, there's no way to do this with Travis, sadly.

Why not setup `./travis.yml` to upload latest nightly DMG artifacts produced by Travis for temporary storing to https://transfer.sh (14 days) or http://0x0.st (30+ days)?

- https://github.com/probonopd/uploadtool
  - https://github.com/probonopd/uploadtool/blob/master/upload.sh#L113-L121

whitequark, 2020-05-14:
Because that's not what CI is for. The goal of continuous integration is to ensure that the codebase builds and passes tests, not to produce artifacts runnable by end users.

The products of Windows and macOS builds aren't particularly suitable for daily work, given that they are debug builds and work *much* slower than the release builds. The products of Linux builds, besides being slow, won't run on an arbitrary Linux system because they're linked against the system dependencies (but it's possible to use Snaps for that).

whitequark, 2020-06-23:
Hi all,

I've identified the [list of issues](https://github.com/solvespace/solvespace/milestone/2) I'd like to see solved in 3.0. This is mostly serious bugs, though a few long-requested enhancements with patches waiting for integration made the list, too.

I'm going to go through that list in the upcoming weeks with the aim to have the 3.0 release out by the end of this summer.

phkahler, 2020-10-20:
All of the issues slated for 3.0 have been closed or pushed back to the next release because they were non-trivial and also not critical. The build infrastructure is being updated and I hope we can have something like a 3.0 beta or even an rc1 very soon.

As for a roadmap, there are a few interesting features under the new 4.0 milestone and that list will likely grow. We still have a great collection of people behind this project so it will progress even though we're all just part-time, so SolveSpace will continue to get better at what it is - The Coolest Little CAD Program on Earth.

closed 2020-10-29 by phkahler