solvespace/solvespace #555: Menu and dialog text shows as squares.

tetious, 2020-03-04:
### System information

SolveSpace version: 3.0~fcb2757d (via snap)

Operating system: Manjaro

### Expected behavior

The menu and dialog text is readable.

### Actual behavior

Menu and dialog text shows as all squares. Feels like a missing font or other wierdness, but not too sure where to start investigating. I'm happy to provide any additional info I can.

The Property Browser is totally normal. This only impacts the menus and dialogs. 

### Additional information

No errors or anything interesting when running via console. 

[image]

ppd, 2020-03-04:
I believe you're seeing that issue: https://forum.snapcraft.io/t/snapped-app-not-loading-fonts-on-fedora-and-arch/12484/30

It's currently under investigation.

tetious, 2020-03-04:
Thanks very much for the pointer to this. The workaround from that thread fixed the problem for me. Gonna close this, as it is clearly not a problem with SolveSpace.

closed 2020-03-04 by tetious