solvespace/solvespace #556: Symbol lookup error: undefined symbol: _glapi_tls_Current

alxv-su, 2020-03-05:
### System information

SolveSpace version: 3.0~fcb2757d

Operating system: Debian 9.12

### Behavior

I get "symbol lookup error: undefined symbol: _glapi_tls_Current" when I try to run solvespace:

```terminal
$ snap run solvespace
/snap/solvespace/16/usr/bin/solvespace: symbol lookup error: /snap/solvespace/16/gnome-platform/usr/lib/x86_64-linux-gnu/libGL.so.1: undefined symbol: _glapi_tls_Current
```

### Additional information

I use proprietary nvidia driver 390.116.

ppd, 2020-03-05:
Is your system glvnd enabled? I.e. do you have stuff in `/var/lib/snapd/lib/glvnd/egl_vendor.d`?

alxv-su, 2020-03-05:
**ppd**, thank you for reply!
My `/var/lib/snapd/lib/glvnd/` directory is empty.
```
root@kitchen:~# snap version
snap    2.43.3
snapd   2.43.3
series  16
debian  9
kernel  4.9.0-12-amd64
```
I have installed all packages that contains `glvnd` (except `nonglvnd`) and `nvidia` in their names from Debian repository. After reboot I still have same error.

ppd, 2020-03-06:
> My `/var/lib/snapd/lib/glvnd/` directory is empty.

Apologies. It's only non-empty from the perspective of the snap. Could you do the following?

```bash
mfederle@maxi:~$ snap run --shell solvespace
To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.

mfederle@maxi:/home/mfederle$ ls -al /var/lib/snapd/lib/glvnd/egl_vendor.d/
total 0
drwxr-xr-x 2 root mfederle 60 Mär  6 09:32 .
drwxrwxrwt 3 root mfederle 60 Mär  6 09:32 ..
lrwxrwxrwx 1 root mfederle 65 Mär  6 09:32 10_nvidia.json -> /var/lib/snapd/hostfs/usr/share/glvnd/egl_vendor.d/10_nvidia.json
```

It should  have somewhat comparable output.

ppd, 2020-03-06:
Do you have `libglvnd0-nvidia` installed? I've seen a few workarounds that mention removing `libglvnd0-nvidia` and installing `libglvnd0`.

alxv-su, 2020-03-06:
I have same file:
```
user@kitchen:~$ snap run --shell solvespace
user@kitchen:/home/user$ ls -al /var/lib/snapd/lib/glvnd/egl_vendor.d
total 0
drwxr-xr-x 2 root user 60 Mar  5 22:38 .
drwxrwxrwt 3 root user 60 Mar  5 22:38 ..
lrwxrwxrwx 1 root user 65 Mar  5 22:38 10_nvidia.json -> /var/lib/snapd/hostfs/usr/share/glvnd/egl_vendor.d/10_nvidia.json
```
Yes, I have `libglvnd0-nvidia` installed.
I'm just replaced `libglvnd0-nvidia` with `libglvnd0`. Now solvespace runs but with text "Unable to create a GL context" in every window.

ppd, 2020-03-06:
Fair enough. So the bug wasn't related to the snap package.

> Now solvespace runs but with text "Unable to create a GL context" in every window.

That's unrelated. What GPU do you have? I think Gtk::GLArea requires >= OpenGL 3.0 features. I wonder if it's possible that your card or driver is too old. But that seems a bit unlikely :thinking:

alxv-su, 2020-03-06:
GeForce GTX 950.
```
$ glxinfo | grep 'OpenGL version'
OpenGL version string: 4.6.0 NVIDIA 390.116
```

ppd, 2020-03-06:
I ran Debian 9 from a USB stick and installed `nvidia-driver` via `stretch-backports`. 

The good news: I can reproduce the issue.
The bad news: It is not specific to solvespace. Other snaps fail just the same with something like:

```shell
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
```

So clearly we're not using the nvidia driver for some reason, yet snapd seems to expose it just fine at a first glance. Something's probably different in the packaging or setup of the Debian driver vs. the Ubuntu one.

ppd, 2020-03-06:
Support request upstream: https://forum.snapcraft.io/t/no-opengl-on-debian-9-nvidia/15848

alxv-su, 2020-03-07:
Is it possible somehow to workaround this problem?
I install Debian 10 on second hard drive and it has same problem.

ppd, 2020-03-07:
You could install the nvidia driver from the official installer or use the nouveau driver.

alxv-su, 2020-03-07:
I just installed nvidia driver 440.64 from the official installer.
Again, solvespace runs but with text "Unable to create a GL context" in every window.

alxv-su, 2020-03-07:
I installed nouveau driver. Solvespace runs, but 3D graphics in other applications is slow.

whitequark, 2020-03-08:
> Is it possible somehow to workaround this problem?

This isn't ideal but building SolveSpace from source should make it work. (Ideally the snap should still be fixed.)

ppd, 2020-03-08:
Or run it without any of the snap confinement:

```shell
mfederle@maxi-xps:~$ cd /snap/solvespace/current
mfederle@maxi-xps:/snap/solvespace/current$ LD_LIBRARY_PATH="$(pwd)/lib:$(pwd)/usr/lib" ./usr/bin/solvespace
SolveSpace!

Missing (absent) translation for '&Helix'
Missing (absent) translation for 'Re&volve'
Missing (absent) translation for 'Show &Underconstrained Points'
^C
mfederle@maxi-xps:/snap/solvespace/current$ 
```

ppd, 2020-03-08:
@whitequark 
> (Ideally the snap should still be fixed.)

The fix will come by update of the snap daemon itself, as it does some magic to expose the nvidia libraries from the host to the app within the snap. We do not ship proprietary drivers in the snap, only the mesa stack, afaik.

ppd, 2020-03-10:
Upstream bug: https://bugs.launchpad.net/snapd/+bug/1866855

alxv-su, 2020-03-10:
> Or run it without any of the snap confinement:
> 
> ```shell
> mfederle@maxi-xps:~$ cd /snap/solvespace/current
> mfederle@maxi-xps:/snap/solvespace/current$ LD_LIBRARY_PATH="$(pwd)/lib:$(pwd)/usr/lib:$(pwd)/usr/lib/x86_64-linux-gnu" ./usr/bin/solvespace
>```
This workaround works! Thank you! :)

closed 2020-03-10 by alxv-su

ppd, 2020-03-18:
For the sake of completeness, I quote upstream's findings:

>  I think right now we have no chance to fix it.

So don't hold your breath; it might take a while...