solvespace/solvespace #559: Build failing on MSYS2 due to zlibstatic and png_static libraries not being found

nabijaczleweli, 2020-03-09:
### System information

SolveSpace version: current HEAD – 6017ecdaccda6bc9ebda1470714e17f8255ea858

Operating system: Windows 10

### Expected behavior

SolveSpace builds.

### Actual behavior

Given the following setup:
  * CMake from the website, and
  * the rest of the dependencies from MSYS2.

The following error occurs:

```
C:/PP/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lzlibstatic
C:/PP/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lpng_static
collect2.exe: error: ld returned 1 exit status
```

The commandline is

```
cd /C/PP/solvespace/build/src && /C/PP/msys64/mingw64/bin/g++.exe  -ffile-prefix-map=C:/PP/solvespace=. -static-libgcc -static-libstdc++ -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Werror=switch   -Wl,--whole-archive CMakeFiles/solvespace-cli.dir/objects.a -Wl,--no-whole-archive  -o ../bin/solvespace-cli.exe -Wl,--out-implib,../bin/libsolvespace-cli.dll.a -Wl,--major-image-version,0,--minor-image-version,0 @CMakeFiles/solvespace-cli.dir/linklibs.rsp
```

With the following in `CMakeFiles/solvespace-cli.dir/linklibs.rsp`:
```
../bin/libsolvespace-core.a ../bin/libsolvespace-headless.a ../bin/libsolvespace-core.a ../bin/libdxfrw.a -lzlibstatic -lpng_static -lfreetype ../bin/libflatbuffers.a ../bin/libcairo.a ../bin/libpixman-1.a -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 
```

When configured for Ninja, the linker arguments are inlined, but to the same effect.

I found these specified on L153, under an `if(WIN32)`:
```cmake
    find_vendored_package(ZLIB zlib
        ZLIB_LIBRARY            zlibstatic
        ZLIB_INCLUDE_DIR        ${CMAKE_SOURCE_DIR}/extlib/zlib)
    list(APPEND ZLIB_INCLUDE_DIR ${CMAKE_BINARY_DIR}/extlib/zlib)

    find_vendored_package(PNG libpng
        SKIP_INSTALL_ALL        ON
        PNG_LIBRARY             png_static
        PNG_PNG_INCLUDE_DIR     ${CMAKE_SOURCE_DIR}/extlib/libpng)
    list(APPEND PNG_PNG_INCLUDE_DIR ${CMAKE_BINARY_DIR}/extlib/libpng)
```

But I can't see any related artifacts in `extlib`:
```
C:\PP\solvespace\build>ls extlib
angle  cairo  flatbuffers  libdxfrw  pixman  q3d
```
Or anywhere at all, `find | grep zlib` and `find | grep png` get nothing while `find .. | grep [lib]` give me just the git module and the source.

I've always assumed this was a problem with my ~~fucked up~~ less than pretty desktop config, and patched `build.ninja` to link to the shared versions I have from MSYS2, but I've reproduced this today on the official [Windows VM image](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/).

whitequark, 2020-03-09:
MSYS2 is not a supported build configuration as per README. Is there a reason you're going for MSYS2 with external dependencies rather than MSVC or MinGW with vendored dependencies? They are vendored specifically to prevent this kind of issue.

whitequark, 2020-03-09:
Wait, I think I misunderstood your setup. You *are* using vendored dependencies, right? Just the compiler and git from MSYS2.

nabijaczleweli, 2020-03-09:
Here's the clean configure output:
```
C:\PP\solvespace\build>cmake -G"MSYS Makefiles" ..
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: C:/PP/msys64/mingw64/bin/gcc.exe
-- Check for working C compiler: C:/PP/msys64/mingw64/bin/gcc.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/PP/msys64/mingw64/bin/g++.exe
-- Check for working CXX compiler: C:/PP/msys64/mingw64/bin/g++.exe - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_FILE_PREFIX_MAP
-- Performing Test HAS_FILE_PREFIX_MAP - Success
-- Using in-tree libdxfrw
-- Using in-tree flatbuffers
-- Looking for strtof_l
-- Looking for strtof_l - not found
-- Looking for strtoull_l
-- Looking for strtoull_l - not found
-- Using in-tree q3d
-- Found ZLIB: C:/PP/msys64/mingw64/lib/libz.dll.a (found version "1.2.11")
-- Found ZLIB: zlibstatic
-- Found PNG: C:/PP/msys64/mingw64/lib/libpng.dll.a (found version "1.6.37")
-- Found Freetype: C:/PP/msys64/mingw64/lib/libfreetype.dll.a (found version "2.10.1")
-- Using in-tree pixman
-- The ASM compiler identification is GNU
-- Found assembler: C:/PP/msys64/mingw64/bin/gcc.exe
-- Building pixman version 0.34.1
-- Performing Test USE_INLINE_inline
-- Performing Test USE_INLINE_inline - Success
-- Looking for getisax
-- Looking for getisax - not found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long
-- Check size of long - done
-- Performing Test HAS_Wall
-- Performing Test HAS_Wall - Success
-- Performing Test HAS_Wdeclaration_after_statement
-- Performing Test HAS_Wdeclaration_after_statement - Success
-- Performing Test HAS_Wno_unused_local_typedefs
-- Performing Test HAS_Wno_unused_local_typedefs - Success
-- Performing Test HAS_fno_strict_aliasing
-- Performing Test HAS_fno_strict_aliasing - Success
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Performing Test USE_LOONGSON_MMI
-- Performing Test USE_LOONGSON_MMI - Failed
-- Performing Test USE_X86_MMX
-- Performing Test USE_X86_MMX - Success
-- Performing Test USE_SSE2
-- Performing Test USE_SSE2 - Success
-- Performing Test USE_SSSE3
-- Performing Test USE_SSSE3 - Success
-- Performing Test USE_VMX
-- Performing Test USE_VMX - Failed
-- Performing Test USE_ARM_SIMD
-- Performing Test USE_ARM_SIMD - Failed
-- Performing Test USE_ARM_NEON
[... 91 lines cut ...]
-- Found ZLIB: zlibstatic (found version "1.2.9")
-- Using in-tree libpng
-- Found ZLIB: zlibstatic
-- Performing Test HAVE_LD_VERSION_SCRIPT
-- Performing Test HAVE_LD_VERSION_SCRIPT - Success
-- Found PNG: png_static (found version "1.6.27")
-- Found Freetype: freetype (found version "2.10.1")
-- Using in-tree pixman
-- Building pixman version 0.34.1
-- Found OpenMP_C: -fopenmp
-- Found OpenMP_CXX: -fopenmp
-- Found OpenMP: TRUE
-- Checking for module 'gtk+-2.0>=2.16'
--   No package 'gtk+-2.0' found
-- Using in-tree cairo
-- Building cairo version 1.15.2
-- Using in-tree ANGLE
-- Benchmarking disabled in debug builds.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/PP/solvespace/build
```

Yes, git and compilers are from MSYS.
And, well, MSYS2 is just what I use on a daily basis, and the `FORCE_VENDORED_*` switches aren't really documented (they exist thrice in the `if(APPLE)` branch and once in FindVendoredPackage), so I rely on CMake getting this right (and it mostly does).

whitequark, 2020-03-09:
Right, OK, indeed the build gets confused about which libz/libpng it should be using, for similar reasons as it did on Apple. I think the right change here would be to force vendored depenencies on Windows always.

nabijaczleweli, 2020-03-09:
It looks like FindVendoredPackage first calls `find_package()` on L18, then overrides it with what it got in varargs – `ZLIB_LIBRARY` and `ZLIB_INCLUDE_DIR` in this case – around L26, and only *then* checks if `find_package()` succeeded on L34.

I don't understand this CMake file completely, but I applied this patch:
```diff
diff --git a/cmake/FindVendoredPackage.cmake b/cmake/FindVendoredPackage.cmake
index 6227f6c..1404740 100644
--- a/cmake/FindVendoredPackage.cmake
+++ b/cmake/FindVendoredPackage.cmake
@@ -16,6 +16,9 @@ include(DisableWarnings)
 function(find_vendored_package PKG_NAME PKG_PATH)
     if(NOT FORCE_VENDORED_${PKG_NAME})
         find_package(${PKG_NAME})
+        if(${VAR_NAME}_FOUND)
+            return()
+        endif()
     endif()
 
     set(cfg_name)
```

Which changed the configure output to:
```
C:\PP\solvespace\build>cmake -G"MSYS Makefiles" ..
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: C:/PP/msys64/mingw64/bin/gcc.exe
-- Check for working C compiler: C:/PP/msys64/mingw64/bin/gcc.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/PP/msys64/mingw64/bin/g++.exe
-- Check for working CXX compiler: C:/PP/msys64/mingw64/bin/g++.exe - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_FILE_PREFIX_MAP
-- Performing Test HAS_FILE_PREFIX_MAP - Success
-- Using in-tree libdxfrw
-- Using in-tree flatbuffers
-- Looking for strtof_l
-- Looking for strtof_l - not found
-- Looking for strtoull_l
-- Looking for strtoull_l - not found
-- Using in-tree q3d
-- Found ZLIB: C:/PP/msys64/mingw64/lib/libz.dll.a (found version "1.2.11")
-- Found ZLIB: C:/PP/msys64/mingw64/lib/libz.dll.a
-- Found PNG: C:/PP/msys64/mingw64/lib/libpng.dll.a (found version "1.6.37")
-- Found Freetype: C:/PP/msys64/mingw64/lib/libfreetype.dll.a (found version "2.10.1")
-- Using in-tree pixman
-- The ASM compiler identification is GNU
-- Found assembler: C:/PP/msys64/mingw64/bin/gcc.exe
-- Building pixman version 0.34.1
-- Performing Test USE_INLINE_inline
-- Performing Test USE_INLINE_inline - Success
-- Looking for getisax
-- Looking for getisax - not found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long
-- Check size of long - done
-- Performing Test HAS_Wall
-- Performing Test HAS_Wall - Success
-- Performing Test HAS_Wdeclaration_after_statement
-- Performing Test HAS_Wdeclaration_after_statement - Success
[... 72 lines cut ...]
-- Check size of void * - done
-- Check size of int
-- Check size of int - done
-- Check size of long long
-- Check size of long long - done
-- Check size of size_t
-- Check size of size_t - done
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for sys/int_types.h
-- Looking for sys/int_types.h - not found
-- Check size of uint64_t
-- Check size of uint64_t - done
-- Check size of uint128_t
-- Check size of uint128_t - failed
-- Check size of __uint128_t
-- Check size of __uint128_t - done
-- Using in-tree ANGLE
-- Benchmarking disabled in debug builds.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/PP/solvespace/build
```

Which looks more promising. I'm running the build now, but it's not terribly fast.

nabijaczleweli, 2020-03-09:
It broke freetype (not unexpectedly), but after forcing *that* to be vendored it correctly linked `solvespace-cli`.

commit e355a095 (2020-03-15): Force vendor zlib, png, and FreeType on Win32, too

This has lead to linker problems if the environment does have standard
versions of zlib/png (like from MSYS2), see discussion in referenced
issue

Closes #559
closed 2020-03-15 by whitequark