solvespace/solvespace #565: Tests crash on i386

nabijaczleweli, 2020-03-23:
### System information

SolveSpace version: current HEAD (e355a095c4c3456487be8bc48675757ab79d44d3)

Operating system: Debian sid i386

### Expected behavior

Tests run

### Actual behavior

```
nabijaczleweli@nabtop:~/src/solvespace/build$ bin/solvespace-testsuite
[...]
test/constraint/pt_face_distance/test.cpp:21: FAILED: render doesn't match reference; 2 (0.00%) pixels differ
  NG   test constraint/pt_face_distance/reference_roundtrip
[...]
File ../test/harness.cpp, line 112, function PrepareSavefile:
Assertion failed: value.size() == newValue.size().
Message: Expected no change in value length.
Backtrace support not compiled in.
Aborted
```

### Additional information

```
nabijaczleweli@nabtop:~/src/solvespace/build$ bin/solvespace-testsuite
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'
  OK   test constraint/length_difference/normal_migrate_from_v22
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
  OK   test constraint/pt_line_distance/free_in_3d_roundtrip
  OK   test core/path/from_raw
  OK   test constraint/proj_pt_distance/reference_migrate_from_v22
  OK   test constraint/where_dragged/free_in_3d_roundtrip
  OK   test constraint/horizontal/line_migrate_from_v22
  OK   test core/path/extension
  OK   test constraint/length_difference/normal_roundtrip
  OK   test constraint/same_orientation/normal_migrate_from_v22
  OK   test constraint/equal_angle/other_roundtrip
  OK   test constraint/symmetric/normal_migrate_from_v22
  OK   test constraint/pt_pt_distance/reference_roundtrip
  OK   test constraint/equal_line_arc_len/tau
  OK   test core/expr/literal
  OK   test constraint/pt_on_line/normal_migrate_from_v22
  OK   test constraint/pt_line_distance/reference_migrate_from_v22
  OK   test constraint/pt_face_distance/normal_migrate_from_v20
  OK   test constraint/angle/normal_roundtrip
  OK   test constraint/diameter/normal_migrate_from_v20
  OK   test constraint/symmetric/free_in_3d_roundtrip
  OK   test constraint/horizontal/pt_pt_roundtrip
  OK   test constraint/length_ratio/normal_roundtrip
  OK   test constraint/pt_line_distance/extended_render
  OK   test constraint/at_midpoint/line_pt_free_in_3d_migrate_from_v22
  OK   test constraint/angle/free_in_3d_roundtrip
  OK   test constraint/cubic_line_tangent/normal_migrate_from_v20
  OK   test constraint/points_coincident/normal_roundtrip
  OK   test constraint/at_midpoint/line_plane_normal_migrate_from_v22
  OK   test constraint/equal_line_arc_len/normal_roundtrip
  OK   test constraint/comment/normal_migrate_from_v20
[... 94 lines cut ...]
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
  OK   test constraint/pt_line_distance/normal_roundtrip
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
test/constraint/pt_face_distance/test.cpp:21: FAILED: render doesn't match reference; 2 (0.00%) pixels differ
  NG   test constraint/pt_face_distance/reference_roundtrip
  OK   test constraint/symmetric_line/normal_migrate_from_v22
  OK   test constraint/arc_line_tangent/normal_migrate_from_v22
  OK   test constraint/proj_pt_distance/reference_roundtrip
  OK   test constraint/pt_pt_distance/reference_migrate_from_v20
  OK   test constraint/symmetric_vert/normal_migrate_from_v22
File ../test/harness.cpp, line 112, function PrepareSavefile:
Assertion failed: value.size() == newValue.size().
Message: Expected no change in value length.
Backtrace support not compiled in.
```

nabijaczleweli, 2020-03-23:
Working diff:
```diff
diff --git a/test/harness.cpp b/test/harness.cpp
index 6012425..6ec319c 100644
--- a/test/harness.cpp
+++ b/test/harness.cpp
@@ -107,7 +107,10 @@ static std::string PrepareSavefile(std::string data) {
                 if(SolveSpaceUI::SAVED[i].fmt  != 'f') continue;
                 if(SolveSpaceUI::SAVED[i].desc != key) continue;
                 double f = strtod(value.c_str(), NULL);
+                fprintf(stderr, "f=%s\n", value.c_str());
+                fprintf(stderr, "  f_pre =%.20f\n", f);
                 f = round(f * precision) / precision;
+                fprintf(stderr, "  f_post=%.20f\n", f);
                 std::string newValue = ssprintf("%.20f", f);
                 ssassert(value.size() == newValue.size(), "Expected no change in value length");
                 std::copy(newValue.begin(), newValue.end(),
```
Relevant value:
```
f=-9.99999999999999822364
  f_pre =-9.99999999999999822364
  f_post=-10.00000000000000000000
File ../test/harness.cpp, line 115, function PrepareSavefile:
```

[Full log (833kB)](https://user-images.githubusercontent.com/6709544/77364659-07479180-6d55-11ea-8cc1-db81e9ceca31.png) (it's plaintext, but GH wouldn't accept other mimetypes sensibly)

whitequark, 2020-03-24:
I think this is because of x87 precision. I suggest simply not using i386, since there is no remaining reason to care about this architecture. But if you want it for some reason, you could try to find the changes to FP environment that bring it more in line with normal double precision floats.

nabijaczleweli, 2020-03-24:
Turns out [passing `-mfpmath=sse -msse2`](https://gcc.gnu.org/wiki/FloatingPointMath) to GCC does solve this; apologies for the clutter and thanks for the pointer!

closed 2020-03-24 by nabijaczleweli

whitequark, 2020-03-24:
We should probably enable that for GCC on i386 in the CMake scripts.

ruevs, 2020-03-24:
If we enable SSE then it is not i386 any more - (it does not have SSE :-).
However it should work if we use i387 (math coprocessor) floating point instructions. They are 80bit doubles and should work fine. I have no GCC machine with me right now to test but from this:

https://gcc.gnu.org/onlinedocs/gcc-3.1/gcc/i386-and-x86-64-Options.html

It appears that we should use:

'-mfpmath=387'

(The reason it is not on by default is that on 80386 the 80387 coprocessor was a separate chip and not every PC had it).

ruevs, 2020-03-24:
https://gcc.gnu.org/wiki/x87note

nabijaczleweli, 2020-03-24:
According to [this](https://gcc.gnu.org/onlinedocs/gcc-9.3.0/gcc/x86-Options.html#x86-Options) GCC doc, `-mfpmath=387` is the default for 32-bit x86 targets (except Darwin, but previous versions of the document don't include that exception), and I've just confirmed this empirically.

Additionally, Clang seems to use SSE2 for floats when allowed to via `-msse2` by default, but fails like GCC otherwise.

ruevs, 2020-03-24:
In this case I think whatever is failing is a bug (I'll take a look when I end up on a PC) and we should not hide it by using SSE.

whitequark, 2020-03-24:
> If we enable SSE then it is not i386 any more - (it does not have SSE :-).

That's incorrect. While you can build software for a target called "i386", no Linux distribution released in the last several years, and indeed no such Linux kernel version, will run without SSE, much less on an actual i386.

> However it should work if we use i387 (math coprocessor) floating point instructions. They are 80bit doubles and should work fine.

Using nonstandard 80 bit floating point numbers is exactly the *cause* of the problem @nabijaczleweli is facing here. They're rounded differently than what the tests are expecting and so the tests don't match.

>  I have no GCC machine with me right now to test but from this:

In the future, please make sure to test your speculation before making authoritative statements. I wrote all of these tests and know perfectly well why they break here.

> In this case I think whatever is failing is a bug (I'll take a look when I end up on a PC) and we should not hide it by using SSE.

The "bug" here is that the testsuite effectively depend on specific rounding modes and floating point sizes. Although I would agree that a testsuite ideally should not depend on that, this was the best I could come up with. Of course, you are welcome to try and do a better job.

ruevs, 2020-03-24:
You are right.