solvespace/solvespace #567: solvespace-cli segfaults in GraphicsWindow::GetCamera()

nabijaczleweli, 2020-03-25:
### System information

SolveSpace version: HEAD (645353cbdc8c0745c9d13b377c28624a61eca15c)

Operating system: Windows 10

### Expected behavior

`solvespace-cli` exports a vector view

### Actual behavior

It segfaults.

### Additional information

I checked back throughout to HEAD~100 (346f004e516b706feb08f9155f69804e4e1d5a47) and got the similar results.

Working diff:
```diff
diff --git a/src/draw.cpp b/src/draw.cpp
index 907ca00..b9139f6 100644
--- a/src/draw.cpp
+++ b/src/draw.cpp
@@ -304,15 +304,25 @@ void GraphicsWindow::GroupSelection() {
 }

 Camera GraphicsWindow::GetCamera() const {
+    fprintf(stderr, "GetCamera(window=%p)\n", window.get());
     Camera camera = {};
+    fprintf(stderr, "Camera\n");
     window->GetContentSize(&camera.width, &camera.height);
+    fprintf(stderr, "GetContentSize\n");
     camera.pixelRatio = window->GetDevicePixelRatio();
+    fprintf(stderr, "pixelRatio\n");
     camera.gridFit    = (window->GetDevicePixelRatio() == 1);
+    fprintf(stderr, "gridFit\n");
     camera.offset     = offset;
+    fprintf(stderr, "offset\n");
     camera.projUp     = projUp;
+    fprintf(stderr, "projUp\n");
     camera.projRight  = projRight;
+    fprintf(stderr, "projRight\n");
     camera.scale      = scale;
+    fprintf(stderr, "scale\n");
     camera.tangent    = SS.CameraTangent();
+    fprintf(stderr, "tangent\n");
     return camera;
 }
```

Result:
```
D:\Programy\solvespace\build>gdb bin\solvespace-cli.exe
GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bin\solvespace-cli.exe...
(gdb) run export-view -o %-2d.svg --view isometric T:\uwu.slvs
Starting program: D:\Programy\solvespace\build\bin\solvespace-cli.exe export-view -o %-2d.svg --view isometric T:\uwu.slvs
[New Thread 41284.0x9ca4]
[New Thread 41284.0x14454]
[New Thread 41284.0xc070]
warning: Missing (absent) translation for group-name'#references'
warning:
Missing (absent) translation for group-name'#references'
warning: Missing (absent) translation for group-name'sketch-in-plane'
warning:
Missing (absent) translation for group-name'sketch-in-plane'
warning: Generate::ALL took 43 ms
warning:
Generate::ALL took 43 ms
warning: Generate::ALL (for bounding box) took 32 ms
warning:
Generate::ALL (for bounding box) took 32 ms
warning: Generate::ALL took 49 ms
warning:
Generate::ALL took 49 ms
GetCamera(window=0000000000000000)
Camera
```

whitequark, 2020-03-25:
I didn't look into this very closely, but it could be a logic bug (null pointer access) that by chance happens to work on Linux. I suggest tracing back the crashing access, it could be that in the CLI, the camera is not filled in for some reason.

nabijaczleweli, 2020-03-25:
Update: doesn't work on Linux, either.

HEAD on buster behaves identically:
```
nabijaczleweli@tarta:~/uwu/solvespace/build$ gdb --args bin/solvespace-cli export-view -o %-2d.svg --view top ../../uwu.slvs
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bin/solvespace-cli...(no debugging symbols found)...done.
(gdb) bt
No stack.
(gdb) run
Starting program: /home/nabijaczleweli/uwu/solvespace/build/bin/solvespace-cli export-view -o %-2d.svg --view top ../../uwu.slvs
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'

Program received signal SIGSEGV, Segmentation fault.
0x000000000041e1fe in SolveSpace::GraphicsWindow::GetCamera() const ()
(gdb) bt
#0  0x000000000041e1fe in SolveSpace::GraphicsWindow::GetCamera() const ()
#1  0x00000000004523cd in SolveSpace::SolveSpaceUI::ExportViewOrWireframeTo(SolveSpace::Platform::Path const&, bool) ()
#2  0x00000000004171e7 in RunCommand(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >)::$_1::operator()(SolveSpace::Platform::Path const&) const ()
#3  0x0000000000417002 in std::_Function_handler<void (SolveSpace::Platform::Path const&), RunCommand(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >)::$_1>::_M_invoke(std::_Any_data const&, SolveSpace::Platform::Path const&) ()
#4  0x000000000041897b in std::function<void (SolveSpace::Platform::Path const&)>::operator()(SolveSpace::Platform::Path const&) const ()
#5  0x0000000000415590 in RunCommand(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) ()
#6  0x0000000000413fdd in main ()
(gdb) quit
```

nabijaczleweli, 2020-03-25:
This comes down to guinone's `CreateWindow()` returning a null pointer (by design, I think, since `Window` is abstract).

<details>
<summary>Working diff</summary>

```diff
diff --git a/src/draw.cpp b/src/draw.cpp
index 907ca00..b9139f6 100644
--- a/src/draw.cpp
+++ b/src/draw.cpp
@@ -304,15 +304,25 @@ void GraphicsWindow::GroupSelection() {
 }
 
 Camera GraphicsWindow::GetCamera() const {
+    fprintf(stderr, "GetCamera(window=%p)\n", window.get());
     Camera camera = {};
+    fprintf(stderr, "Camera\n");
     window->GetContentSize(&camera.width, &camera.height);
+    fprintf(stderr, "GetContentSize\n");
     camera.pixelRatio = window->GetDevicePixelRatio();
+    fprintf(stderr, "pixelRatio\n");
     camera.gridFit    = (window->GetDevicePixelRatio() == 1);
+    fprintf(stderr, "gridFit\n");
     camera.offset     = offset;
+    fprintf(stderr, "offset\n");
     camera.projUp     = projUp;
+    fprintf(stderr, "projUp\n");
     camera.projRight  = projRight;
+    fprintf(stderr, "projRight\n");
     camera.scale      = scale;
+    fprintf(stderr, "scale\n");
     camera.tangent    = SS.CameraTangent();
+    fprintf(stderr, "tangent\n");
     return camera;
 }
 
diff --git a/src/graphicswin.cpp b/src/graphicswin.cpp
index 0d520dd..4e08553 100644
--- a/src/graphicswin.cpp
+++ b/src/graphicswin.cpp
@@ -404,7 +404,9 @@ void GraphicsWindow::Init() {
     toolbarHovered = Command::NONE;
 
     if(!window) {
+        printf("GW::Init(no window)\n");
         window = Platform::CreateWindow();
+        printf("CreateWindow() = %p\n", window.get());
         if(window) {
             using namespace std::placeholders;
             // Do this first, so that if it causes an onRender event we don't try to paint without
diff --git a/src/platform/entrycli.cpp b/src/platform/entrycli.cpp
index ad0665d..9b252ab 100644
--- a/src/platform/entrycli.cpp
+++ b/src/platform/entrycli.cpp
@@ -355,6 +355,7 @@ static bool RunCommand(const std::vector<std::string> args) {
         Platform::Path absOutputFile = outputFile.Expand(/*fromCurrentDirectory=*/true);
 
         SS.Init();
+        printf("SS.Init(SS.GW.window=%p)\n", SS.GW.window.get());
         if(!SS.LoadFromFile(absInputFile)) {
             fprintf(stderr, "Cannot load '%s'!\n", inputFile.raw.c_str());
             return false;
diff --git a/src/platform/guigtk.cpp b/src/platform/guigtk.cpp
index 73fb01e..968be25 100644
--- a/src/platform/guigtk.cpp
+++ b/src/platform/guigtk.cpp
@@ -1022,6 +1022,7 @@ public:
 
 WindowRef CreateWindow(Window::Kind kind, WindowRef parentWindow) {
     auto window = std::make_shared<WindowImplGtk>(kind);
[... 135 lines cut ...]
CreateWindowNone((nil))
CreateWindow() = (nil)
CreateWindowNone((nil))
SS.Init(SS.GW.window=(nil))
GW::Init(no window)
CreateWindowNone((nil))
CreateWindow() = (nil)
CreateWindowNone((nil))
GetCamera(window=(nil))
Camera

Program received signal SIGSEGV, Segmentation fault.
0x000000000041e297 in SolveSpace::GraphicsWindow::GetCamera() const ()
(gdb) bt
#0  0x000000000041e297 in SolveSpace::GraphicsWindow::GetCamera() const ()
#1  0x000000000045253d in SolveSpace::SolveSpaceUI::ExportViewOrWireframeTo(SolveSpace::Platform::Path const&, bool) ()
#2  0x0000000000417217 in RunCommand(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >)::$_1::operator()(SolveSpace::Platform::Path const&) const ()
#3  0x0000000000417032 in std::_Function_handler<void (SolveSpace::Platform::Path const&), RunCommand(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >)::$_1>::_M_invoke(std::_Any_data const&, SolveSpace::Platform::Path const&) ()
#4  0x00000000004189cb in std::function<void (SolveSpace::Platform::Path const&)>::operator()(SolveSpace::Platform::Path const&) const ()
#5  0x00000000004155c0 in RunCommand(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) ()
#6  0x0000000000413fed in main ()
(gdb) quit
```

</details>

phkahler, 2020-10-02:
Did you two ever get any further on this? Unfortunately the CLI works fine here on Fedora.

nabijaczleweli, 2020-10-03:
That analysis is as far as I've gotten. I'll be able to test again after the week-end.

nabijaczleweli, 2020-10-05:
At c514ddad5402e8ca53d19948f33dcc397e58ae9b, current HEAD:
```
D:\Programy\solvespace\build>gdb --args bin\solvespace-cli.exe export-view -o %-2d.svg --view top P:\Embedded\b_nch\top.slvs
GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bin\solvespace-cli.exe...
(gdb) run
Starting program: D:\Programy\solvespace\build\bin\solvespace-cli.exe export-view -o %-2d.svg --view top P:\Embedded\b_nch\top.slvs
[New Thread 10652.0x2e38]
[New Thread 10652.0x2e44]
[New Thread 10652.0x3074]
warning: Missing (absent) translation for group-name'#references'
warning:
warning: Missing (absent) translation for group-name'sketch-in-plane'
warning:
warning: Generate::ALL took 45 ms
warning:

Thread 1 received signal SIGSEGV, Segmentation fault.
0x0000000000405fd6 in SolveSpace::GraphicsWindow::GetCamera (this=this@entry=0x994a10 <SolveSpace::SS+16>)
    at D:/Programy/Programowanie/msys2/mingw64/include/c++/9.3.0/bits/shared_ptr_base.h:1020
1020          _M_get() const noexcept
(gdb) bt
#0  0x0000000000405fd6 in SolveSpace::GraphicsWindow::GetCamera (this=this@entry=0x994a10 <SolveSpace::SS+16>)
    at D:/Programy/Programowanie/msys2/mingw64/include/c++/9.3.0/bits/shared_ptr_base.h:1020
#1  0x000000000042f5cb in SolveSpace::SolveSpaceUI::ExportViewOrWireframeTo (this=0x994a00 <SolveSpace::SS>, filename=..., exportWireframe=<optimized out>)
    at ../src/export.cpp:234
#2  0x0000000000402f23 in std::function<void (SolveSpace::Platform::Path const&)>::operator()(SolveSpace::Platform::Path const&) const (__args#0=..., this=0x84cfc70)
    at D:/Programy/Programowanie/msys2/mingw64/include/c++/9.3.0/bits/std_function.h:683
#3  RunCommand (args=...) at ../src/platform/entrycli.cpp:364
#4  0x0000000000849464 in main (argc=7, argv=0x194880) at ../src/platform/entrycli.cpp:382
```

nabijaczleweli, 2020-10-05:
Adding
```diff
diff --git a/src/draw.cpp b/src/draw.cpp
index db2f33e..f40b80f 100644
--- a/src/draw.cpp
+++ b/src/draw.cpp
@@ -304,6 +304,7 @@ void GraphicsWindow::GroupSelection() {
 }
 
 Camera GraphicsWindow::GetCamera() const {
+    fprintf(stderr, "GetCamera(window=%p)\n", window.get());
     Camera camera = {};
     window->GetContentSize(&camera.width, &camera.height);
     camera.pixelRatio = window->GetDevicePixelRatio();
```
Like in the first diff prints "GetCamera(window=0000000000000000)", so that's that

ruevs, 2020-10-16:
`window` is  `Platform::WindowRef   window;` which is  `typedef std::shared_ptr<Window> WindowRef;`
```C++
Camera GraphicsWindow::GetCamera() const {
    Camera camera = {};
    window->GetContentSize(&camera.width, &camera.height);
```
```
Unhandled exception thrown: read access violation.
std::shared_ptr  (...) returned nullptr.
```
This happens because `window` indeed ends up being a `nullptr` because `GraphicsWindow::Init()` does:
https://github.com/solvespace/solvespace/blob/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b/src/graphicswin.cpp#L413
which ends up here:
https://github.com/solvespace/solvespace/blob/d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b/src/platform/guinone.cpp#L95

What is more the virtual `Window` class is not even overloaded in guinone.cpp so things like `GetContentSize` do not exist in `solvespace-cli`!

I wrote a fix that allows 
`solvespace-cli export-view -o %-2d.svg --view front something.slvs`
to run and it create a SVG file.

closed 2020-10-16 by phkahler