solvespace/solvespace #569: Add midpoint constraint

Timmmm, 2020-03-28:
SolveSpace version: 3.0~645353cb (master from a week or so ago)
Operating system: Windows 10

It would be really nice if there were a "midpoint on line" constraint like in other CAD software (pretty sure Solidworks has this). Basically you select a line, and a point, and the point is constrained to be at the midpoint of the line.

Alternatively maybe every line could have a "virtual" midpoint, which is like a normal point but it can't be deleted. (I guess this is kind of how circle/arc centres and bezier control points work?)

Evil-Spirit, 2020-03-28:
Hello, just select point and line then press "M"

Timmmm, 2020-03-28:
Ah yeah that works! I thought the constraints in the toolbar were all of them - I'll add a task to add the missing ones.

closed 2020-03-28 by Timmmm